SolderedElectronics/Inkplate-Arduino-library #8: Check Adafruit License

blazer82, 2020-07-26:
It seems you copied the Adafruit GFX library into your project without its license notice. The Adafruit library is licensed under the two-clause BSD license which should give you the freedom to use its source if done correctly. 

Alternatively, have you checked whether you can just add the Adafruit library as a dependency to your library? That way you wouldn’t need to copy its code. That’s what Adafruit does with their libraries that build on the GFX library.

davidzovko, 2020-07-27:
Hello! I appreciate your comment. Adafruit GFX was added directly to our repo for two reasons:
- simplicity for users: just install one library and everything works
- if GFX changes in some critical part, we will have time to adjust Inkplate library to work with it and avoid temporary incompatibility between the two 

As for the license, thanks again. Can you please tell us, if you know, where exactly the licence notice should be put?

blazer82, 2020-07-28:
Thanks for the reply. I see now why you copied it. I'm not an expert on licensing, so I'm not exactly sure how to go from here. Maybe move the Adafruit stuff into its own directory and also place Adafruit's license file in it.

Or just ask Adafruit what they'd prefer.

Really cool project, by the way!

davidzovko, 2020-07-28:
Thanks anyway, I'll look into licensing and contact Adafruit if needed!

martinberlin, 2020-07-28:
I think that's appropiate. At least that's how I see it's done on [other projects](https://github.com/martinberlin/cale-idf/tree/master/components). By the way I would give it a try to make an ESP-IDF Esp32 component for Espressif own framework if I find the time for it.

davidzovko, 2020-07-28:
ESP-IDF ESP32 component would be terrific add-on ;) if you decide to do such thing, feel free to contact us on the email!

davidzovko, 2020-07-31:
After communication with Adafruit, license as per their instruction has been added.

closed 2020-07-31 by davidzovko