SolderedElectronics/Inkplate-Arduino-library #10: Include the shipping demo?

wohali, 2020-07-28:
Hello there, my Inkplate shipped with an application showing: https://www.crowdsupply.com/img/5de4/inkplate-in-hand-edited_jpg_project-body.jpg (except also showing "Slave mode Ready" and the Panel VCOM Voltage.

Is this example available anywhere? It doesn't seem to be identical to any of the shipping examples.

davidzovko, 2020-07-28:
Hey, every Inkplate 6 is shipped with Slave Mode software preloaded on it. It enables you to use Inkplate 6 without reprogramming, via UART commands. You can find the slave mode code here: https://github.com/e-radionicacom/Inkplate-6-Arduino-library/tree/master/examples/3.%20Others/1-Inkplate_Slave_Mode

In the programming code, which result is the one you can see on the photo you have attached, we write VCOM voltage to TPS power supply chip. This shouldn't be done too often and that's why the code hasn't been published. We also need to polish one detail so it is done 100% automatically. Other than that, the code is identical to Slave mode example from the library. 

Since there's interest, we will post the code when this one error is fixed.

davidzovko, 2020-07-31:
If there's no more questions, we'll close those issue.

wohali, 2020-07-31:
Hi @davidzovko , thanks for your attention. I am  hoping the _exact_ code load that is on the shipping unit (with the Thank you for purchasing... text and picture, slave mode, and display of the battery) could be uploaded and provided.  I'm happy to wait until the voltage write bug is fixed.

davidzovko, 2020-08-02:
I see! I will leave the issue open until we provide it, in that case.

davidzovko, 2020-12-07:
Added, at last! 

https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/3bc9c5879686f5a3ccbfe532e1f5beec80f7d45b

closed 2020-12-07 by davidzovko