SolderedElectronics/Inkplate-Arduino-library #11: Could not load any BMP image on slave peripherical mode

martinberlin, 2020-07-28:
Tried only storing 1.bmp, 2.bmp and 3.bmp that I converted to 4 bits per pixel and stored them in the SD card.

So then I just inserted it. And tried sending: 

#H(000,000,"/1.bmp")*

And I get always:
#H(0)* - Image load failed as response

Is there any special requirements for the BMP bitdepth and so on? Could you add it also into the documentation please.
To reproduce:
[inkscape.zip](https://github.com/e-radionicacom/Inkplate-6-Arduino-library/files/4987770/inkscape.zip) -> 1 & 2 attached.

Another commands like fillCircle work.
Could not send #C("DEMO")*
This prints for me just garbage on the screen but not any readable text. Will try again with setCursor to see if it was a positioning issue.

davidzovko, 2020-07-28:
Hey Martin! 

Sorry, some documentation is available [here](https://github.com/e-radionicacom/Inkplate-6-Arduino-library/blob/master/examples/3.%20Others/1-Inkplate_Slave_Mode/Inkplate_slave_mode_documentation.txt) as you already know. We are working on writing the proper one (it's a lot of work!). 
We are looking into this error with SD card and will let you know ASAP.

As for #C("STRING")* command which should print text, please check this note:
`STRING - Text/Strig coded in HEX Char (example: HELLO WORLD would be coded like 48454c4c4f20574f524c44, where 48 means 0x48 which is H in ASCII table)`
so you'll have to encode it to HEX to get the readable text

BornaBiro, 2020-07-28:
Same goes for loading images from microSD card using slave mode.
Instead of sending command #H(000,000,"/1.bmp")* you should send #H(000,000,"2f312e626d70")* where 2f312e626d70 is /1.bmp coded in hex.
For conversion from string to hex char you can use this handy online converter: https://codebeautify.org/string-hex-converter

martinberlin, 2020-07-28:
Thanks. But I must say this is not really straight forward!
You should simply decode every char to their HEX counterpart. Here [an example](https://stackoverflow.com/questions/3381614/c-convert-string-to-hexadecimal-and-vice-versa).

davidzovko, 2020-07-28:
It's valid idea - but we force HEX for the edge case where your string could contain a slave mode command which would lead to malfunction. For example: #H(000,000,"/image#L(1)*.bmp")* - here you would get a screen refresh before it is time for one.

davidzovko, 2020-07-28:
If you have other idea how to handle this, feel free to let us know!

martinberlin, 2020-07-28:
I would simply add every char after the " to their HEX counterpart. Well if the user inserts a #L(1) or whatever they will just get rendered that in the screeen. 
But at least then it is more easier to use (And matches what is specified in the [documentation](https://github.com/e-radionicacom/Inkplate-6-Arduino-library/blob/master/examples/3.%20Others/1-Inkplate_Slave_Mode/Inkplate_slave_mode_documentation.txt#L134))
Let me see if I can find time to check how the paralell communication is done. I guess "slave" option is more like a demo to try the epaper when it arrives, probably not worth to spend too much time, but still some small optimizations may add up to make it easier to use. 
The most easy way to send serial like this is to use Arduino IDE.

davidzovko, 2020-07-29:
Thanks on all your suggestions, Martin! We'll reconsider and check what can be done! ([documentation](https://github.com/e-radionicacom/Inkplate-6-Arduino-library/blob/master/examples/4.%20Others/1-Inkplate_Slave_Mode/Inkplate_slave_mode_documentation.txt) has been updated btw!)

Slave mode is actually intended for people that are used to their microcontroller/ecosystem and just want to power up Inkplate as quick as possible. You can use it with Raspberry Pi or some other microcontroller with just three lines (TX, RX, gnd). 
Parallel is way to go, but there's no hardware capabilities on the Inkplate -- and might be overkill for users that want simplicity. The UART & parallel could be a good combo, though.

martinberlin, 2020-07-29:
Nice. Just let me know if it does not work the string to HEX conversion and I will clone it and make a pull request!

closed 2020-07-29 by martinberlin