SolderedElectronics/Inkplate-Arduino-library #16: Compiling this repository with Platformio

martinberlin, 2020-07-30:
Since long time I don't use Arduino IDE so I tried to compile this repository using https://platformio.org
My platformio.ini configuration file is the following:

```
[platformio]
default_envs = esp32

[env:esp32]
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
monitor_speed = 115200
board = esp-wrover-kit
; CPU Frequency. Max for ESP32: 240000000L
board_build.f_cpu = 240000000L
lib_deps=
   https://github.com/e-radionicacom/Inkplate-6-Arduino-library.git
   https://github.com/e-radionicacom/Inkplate-6-SDFat-Arduino-Library.git
```
It compiles with just a few yellow warnings (Not important now)
board is not exactly the right one, but in the [boards](https://docs.platformio.org/en/latest/boards/index.html) section, is not listed esp-wrover (only esp-wrover-kit) but I don't think that's a problem anyways. 

As for the sketch, just trying to clear the screen, but something strange happens:

```c++
#include "Inkplate.h"               //Include Inkplate library to the sketch
#include "HTTPClient.h"             //Include library for HTTPClient
#include "WiFi.h"                   //Include library for WiFi
Inkplate display(INKPLATE_1BIT);   

const char* ssid     = "KabelBox-A210"; //Your WiFi SSID
const char* password = "xxx";     //Your WiFi password

void setup() {
    Serial.begin(115200);delay(19);
    Serial.println("Hello inkplate");
  display.begin();        //Init Inkplate library (you should call this function ONLY ONCE)
  Serial.println("display.begin()");  // --> It stops just after display.begin() I don't see this line already
  display.clearDisplay(); //Clear frame buffer of display
  Serial.println("display.clearDisplay()");
}

void loop() {
  //Nothing...
}
```

I am missing something?
I just wanted to check how Inkplate works against my [epaper service web-service](https://cale.es). By the way I added the Inkplate6 as an epaper in the database if someone wants to try it:
https://cale.es/eink-displays

BornaBiro, 2020-07-30:
Reason why the program stops in display.begin() is because dynamic memory allocation at PSRAM has failed. The reason why it failed can be:
 - You don't have enough RAM available for the frame buffer
 - External PSRAM of ESP32 is not enabled.

As you said, board is not exacly right one and if is ESP32 selected instead ESP32 WROVER, that can be the cause PSRAM not activating. And as result code stops at display.begin().

In arduino IDE there is option, if ESP32 Dev Board is selected, you can activate PSRAM, by selecting it to ENABLE instead of default DISABLE.

martinberlin, 2020-07-30:
Yes true. Then Board configuration is the key. But is strange you don't get any error message to make you aware of this ;)
So I'm adding a custom board based on an existing one that has 4Mb PSRAM. Thanks I will report here when I resolve it in case it can help someone

martinberlin, 2020-07-31:
Resolved. If anyone wants to test compiling this on Platformio I left here a repository ready to be tested:

https://github.com/martinberlin/inkplate-platformio

One thing, the example:
**examples/2. Advanced Inkplate Features/10-Inkplate_Download_And_Show** 

Is not compiling:
error: 'class Inkplate' has no member named 'drawBitmapFromWeb'
   if(!display.drawBitmapFromWeb("https://varipass.org/neowise_mono.bmp", 0, 0, true)) {

And is true. Inkplate class has no method called drawBitmapFromWeb

Can you please correct this?

davidzovko, 2020-07-31:
Sorry about drawBitmapFromWeb error - please try now.

Will look into platformio later!

martinberlin, 2020-07-31:
Thanks now it compiles, could not test it with Cale since I don't have internet in my mobile now ;)
Feel free to close this after testing my Platformio take. 
What I like about VSCode IDE and Platformio as an extension, is that you just download one repository, and it installs dependencies automatically. And you have build_flags that you can set in the platformio.ini file (For example WiFi credentials) keeping all really tight together and easy to compile. 
Maybe someone else will find it useful to quick start with Inkplate.

svwhisper, 2020-08-01:
Thank you for getting this working.  Platformio is the only way to go for complex code.

martinberlin, 2020-09-04:
Working correctly on platformio

closed 2020-09-04 by martinberlin

Gearme, 2020-09-12:
> Thanks now it compiles, could not test it with Cale since I don't have internet in my mobile now ;)
> Feel free to close this after testing my Platformio take.

I'm sure I must have missed something, but it seems to me like master (which both of you are referencing in your repos) does _not_ have drawBitmapFromWeb. Only the revision branch does, but that one causes other errors for me. Were you testing with that one?

martinberlin, 2020-09-13:
If I recall correctly is already merged in master.
Just clone:
https://github.com/martinberlin/inkplate-platformio

Or try to delete the piolibs/library folder so.it downloads a fresh copie. It should be there since I could render the image.

Gearme, 2020-09-13:
My copy is fresh, no such luck. But I couldn't find signature or implementation, so I went through the commits - looks like David (accidentally?) removed it again.

https://github.com/e-radionicacom/Inkplate-6-Arduino-library/commit/c9c2031c66e106bab495a93e6137e25fd385ea45

davidzovko, 2020-09-15:
We are working on a library revision - so those are just temporarily removed. You can find the _drawBitmapFromWeb_ in "dev" branch: https://github.com/e-radionicacom/Inkplate-6-Arduino-library/tree/dev

In a week time, those should be back to the master branch.