SolderedElectronics/Inkplate-Arduino-library #17: Minor clarifications to getting started directions

skybrian, 2020-07-31:
First, the directions say:

> Optional: if you are planning to use the SD card functionality, install custom SdFat library into your Arduino IDE.

However, Inkplate.h includes Sdfat.h unconditionally, so as the code currently stands, it's not actually optional. I used "git clone" to install the SDFat library and that fixed it.

Second, as shipped, with the optional cover on, you can't tell which way to push the power switch to turn it on, because the screen doesn't change and there's no light. "On" is away from from the USB jack. If the device is off, the serial port still appears under /dev/tty.wchusbserial* when you plug in the USB jack (assuming you have the drivers installed correctly), but uploading fails with a timeout.

After that I was able to install the Inkplate_basic_monochrome example.

davidzovko, 2020-07-31:
Thank you for the tip @skybrian, that is correct, Sdfat.h is needed and we have updated the instructions. 

It's a good tip, we could have added the power indication embossed into 3D printed case. Will be implemented in future versions!

closed 2020-07-31 by davidzovko

davidzovko, 2020-07-31:
Added: https://github.com/e-radionicacom/Inkplate-6-hardware/pull/6