SolderedElectronics/Inkplate-Arduino-library #21: bits/c++allocator.h: No such file or directory

Peter-B-, 2020-08-13:
Hi,

I just got my Inkplate and tried to get started, but I cannot compile the examples.

I followed the instruction and installed `Croduino ESP32 Boards 1.0.0`. I installed this repository as a library from the `.zip` file. When I open any of the examples (even `Inkplate_Clean_Screen.ino`), I see the following error:

```
In file included from c:\users\peter.butzhammer\appdata\local\arduino15\packages\croduino_boards\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\string:41:0,
                 from c:\users\peter.butzhammer\appdata\local\arduino15\packages\croduino_boards\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\random:40,
                 from c:\users\peter.butzhammer\appdata\local\arduino15\packages\croduino_boards\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:66,
                 from c:\users\peter.butzhammer\appdata\local\arduino15\packages\croduino_boards\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62,
                 from C:\Users\peter.butzhammer\AppData\Local\Arduino15\packages\Croduino_Boards\hardware\esp32\1.0.0\cores\esp32/Arduino.h:148,
                 from sketch\2-Inkplate_Clean_Screen.ino.cpp:1:
c:\users\peter.butzhammer\appdata\local\arduino15\packages\croduino_boards\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\allocator.h:46:75: fatal error: bits/c++allocator.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Inkplate 6(ESP32).
```

And there is indeed no `bits/c++allocator.h` file or `bits` subdirectory.

Do you have any idea what might be wrong?

Greetings
Peter

davidzovko, 2020-08-26:
Hi Peter,

sorry for the issue and long time needed to reply. We were having some time off. 
Looks like the installation of board definition went wrong. That file should exist. 

Can you try the following? Delete these folders:
`C:\documents and settings\USERNAME\local settings\data aplikací\arduino15\packages\esp8266`
`c:\documents and settings\USERNAME\local settings\data aplikací\arduino15\staging\packages`

Close arduino and run it again. 
Reinstall board definition. 
Try compiling again.

Did that help?

Peter-B-, 2020-08-26:
Hi David, 

thanks for your reply. 

My `arduino 15\packages` folder does not contain an `esp8266` directory. Maybe this is the problem.

I just uninstalled Arduino, deleted the whole `Arduino 15` folder and reinstalled the latest version of Arduino and the Inkplate board, but that did not fix my problem. 

I will try to investigate the problem in the next days.

Greetings
Peter

davidzovko, 2020-08-27:
Sorry, my mistake. The folder in `arduino 15\packages` should be `esp32`. I typed incorrectly. 

But, if you uninstalled and deleted whole `arduino 15` folder, that should have fixed it, you deleted the `esp32` folder right with it. 

A little hard for me to debug since I can't replicate the problem. Does file `c:\users\peter.butzhammer\appdata\local\arduino15\packages\croduino_boards\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\allocator.h` actually exist after you install the board definition for Inkplate?

Peter-B-, 2020-08-27:
Today, tested it on another computer and there it seems to work. 

Thanks for your efforts.

I will try to copy the files from one PC to the other. I am curious if this will solve the problem there.

closed 2020-08-27 by Peter-B-

davidzovko, 2020-08-28:
That's interesting! If you find something out, please let us know!