SolderedElectronics/Inkplate-Arduino-library #25: Commit a1578be breaks my example for partial refresh

joerg-d, 2020-08-30:
From commit a1578be onwards [my example code](https://github.com/joerg-d/Inkplate-6-Arduino-library/tree/deep-sleep-partial-refresh/examples/2.%20Advanced%20Inkplate%20Features/11-Inkplate_Low_Power_Partial_Update) produces garbage after partial refresh. Some parts that have to clear stay black. Unmodified parts are grayed out.

davidzovko, 2020-08-30:
Can you please post your code here? We just tested with the library from a1578be and stock example code, no problems.

joerg-d, 2020-08-30:
My simple clock example:

```cpp
#include "Inkplate.h"                  //Include Inkplate library to the sketch
#include "driver/rtc_io.h"             //ESP32 library used for deep sleep and RTC wake up pins
#include "driver/adc.h"
#include <esp_wifi.h>
#include <esp_bt.h>

//Including bitmaps (created with the LCD Image Converter)
typedef struct {
    const uint8_t *data;
    uint16_t width;
    uint16_t height;
    uint8_t dataSize;
    } tImage;
#include "binary_icons/icon_char_colon.h"
#include "binary_icons/icon_digit_0.h"
#include "binary_icons/icon_digit_1.h"
#include "binary_icons/icon_digit_2.h"
#include "binary_icons/icon_digit_3.h"
#include "binary_icons/icon_digit_4.h"
#include "binary_icons/icon_digit_5.h"
#include "binary_icons/icon_digit_6.h"
#include "binary_icons/icon_digit_7.h"
#include "binary_icons/icon_digit_8.h"
#include "binary_icons/icon_digit_9.h"
const uint8_t *digits[10] ={ image_data_icon_digit_0, image_data_icon_digit_1, image_data_icon_digit_2, image_data_icon_digit_3, image_data_icon_digit_4, image_data_icon_digit_5, image_data_icon_digit_6, image_data_icon_digit_7, image_data_icon_digit_8, image_data_icon_digit_9 };

//Includes
#include <ctime>
#include "Network.h"

//CHANGE HERE ---------------

char *ssid = "Freifunk";
char *pass = "";
int timeZone = 2;

//Set to 1 to flip the screen 180 degrees
#define ROTATION 3

//---------------------------

#define uS_TO_S_FACTOR 1000000         //Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  60              //How long ESP32 will be in deep sleep (in seconds)

RTC_DATA_ATTR bool firstboot = true;
RTC_DATA_ATTR int timeDigits[4];
RTC_DATA_ATTR int wakeups = 0;
const int refreshesToGet = 60;         // full refresh every hour
const int refreshesToSetTime = 60*24;  // get time update once every day

Inkplate display(INKPLATE_1BIT);       //Create an object on Inkplate library and also set library into 1 Bit mode
Network network;                       //Our networking functions, see Network.cpp for info

//All our functions declared below setup and loop
void drawTime();

void setup() {

    //Initial display settings
    display.begin();                   //Before calling any display method you must call .begin()
    display.clearDisplay();            //Clears all data in buffer.
    display.setRotation(ROTATION);

    if (firstboot)
        firstboot = false;
    else {
        //restore old bitmap (before the deep sleep)
        drawTime();
        display.load1b();
    }

    //Update data
    if (wakeups % refreshesToSetTime == 0) {
        network.begin();
        network.off();
    }
    network.getTime(timeDigits);

    // create new bitmap
    display.clearDisplay();
    drawTime();

    //Actually display all data
    if (wakeups % refreshesToGet == 0) {
        display.clean();               //Cleans the actual screen of any possible burn in.
        display.display();
    }
    else
        display.partialUpdate();
    ++wakeups;

    //Sleep
    adc_power_off();                                                  //don't forget to turn ADC back before using it (also after wake up from deep sleep)
    esp_wifi_stop();
    esp_bt_controller_disable();
    rtc_gpio_isolate(GPIO_NUM_12);                                    //Isolate/disable GPIO12 on ESP32 (only to reduce power consumption in sleep)
    esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);    //Activate wake-up timer -- wake up after 20s here
    esp_deep_sleep_start();                                           //Put ESP32 into deep sleep. Program stops here.
}

void loop() {
  //Nothing! If you use deep sleep, whole program should be in setup() because each time the board restarts, not in a loop()! loop() must be empty!

}

//Drawing what time it is
void drawTime()
{
    display.drawBitmap(  0, 0, digits[timeDigits[3]],     134, 317, WHITE, BLACK); //hours (tens)
    display.drawBitmap(134, 0, digits[timeDigits[2]],     134, 317, WHITE, BLACK); //hours (units)
    display.drawBitmap(268, 0, image_data_icon_char_colon, 64, 317, WHITE, BLACK); //colon
    display.drawBitmap(332, 0, digits[timeDigits[1]],     134, 317, WHITE, BLACK); //minutes (tens)
    display.drawBitmap(466, 0, digits[timeDigits[0]],     134, 317, WHITE, BLACK); //minutes (units)
}
```

The function "load1b" is added to Inkplate.cpp:
```cpp
void Inkplate::load1b()
{
    for (int i = 0; i < 60000; i++)
    {
        *(D_memory_new + i) &= *(_partial + i);
        *(D_memory_new + i) |= (*(_partial + i));
    }
    _blockPartial = 0;
}
```

BornaBiro, 2020-08-30:
Partial updates should not work with deep sleep.
The way partial updates are currently working is that we have two frame buffers: one that stores everything that is currently on the screen and one that stores everything that needs to be written on the screen in the next screen update. In partial updates, code will only look up for the differences between these two buffers and only send that to the screen. The problem is that deep sleep disables the RAM (internal and external one in ESP32 module) and erases everything form it, so the code does not have anything to compare with and sends garbage.
There is a safety mechanism that should prevent this from happening (it only allows partial updates if the screen is at least once fully refreshed/updated) with _blockPartial variable.

joerg-d, 2020-08-30:
Yes, I know, partial updates should not work with deep sleep. Because of this, my code tries to recreate the state before the deep sleep: It draws the same bitmaps to the first frame buffer. Instead of calling display.display() to write this to the second frame buffer and update the e-paper with flickering, my code only write the content from the first to the second frame buffer. The e-paper doesn't need updating, it shows the same content already. After this I update the first frame buffer with new data and call display.partialUpdate(). Therefore I have no flickering after deep sleep. This is working without any problems until a1578be.

davidzovko, 2020-09-01:
Hey, we understand now. Since it is a library modification, could we kindly ask you to post that on our forum and we'll take a look there, since this is not issue related to the library. 

Forum: http://forum.e-radionica.com/en/viewforum.php?f=22&sid=04537e0b2c8ff42d91f3e0e813ef92c8

joerg-d, 2020-09-01:
Okay, this is the forum post: http://forum.e-radionica.com/en/viewtopic.php?f=22&t=274

closed 2020-09-01 by joerg-d