SolderedElectronics/Inkplate-Arduino-library #28: no matching function for call to 'SdFat::SdFat(SPIClass*)'

achill-es, 2020-08-30:
Hi,

when I'm trying to compile "2-Hourly_weather_station_example" on my Windows 10 Arduino IDE 1.8.13, I receive this error message:

-----
C:\Users\achill-es\OneDrive\Arduino\libraries\Inkplate-6-Arduino-library-master\Inkplate.cpp:8:15: error: no matching function for call to 'SdFat::SdFat(SPIClass*)'
 SdFat sd(&spi2);
               ^
In file included from C:\Users\achill-es\OneDrive\Arduino\libraries\Inkplate-6-Arduino-library-master\Inkplate.h:16:0,
                 from C:\Users\achill-es\OneDrive\Arduino\libraries\Inkplate-6-Arduino-library-master\Inkplate.cpp:6:
-----

Btw: Of course, I'm using the latest versions of all Inkplate libraries (incl. "Inkplate-6-SDFat-Arduino-Library-master").

nitko12, 2020-08-30:
Hi, 
that's weird, as the example doesn't use the SD explicitly.
Do other examples compile?

achill-es, 2020-08-30:
They did 2 weeks ago. But stopped today for unknown reason.

achill-es, 2020-08-31:
Okay, I maybe spotted it: As soon as I deinstalled any SdFat library apart from "Inkplate-6-SDFat-Arduino-Library-master", it compiled again smoothly. Guess you should tell people there's a conflict and they should think of "Inkplate-6-SDFat-Arduino-Library-master" being a replacement for any other SdFat.

davidzovko, 2020-09-01:
That is correct, only Inkplate version of SdFat should be installed. We will look into how to make this workable.

closed 2020-09-01 by davidzovko