SolderedElectronics/Inkplate-Arduino-library #32: DrawBitmapFromWeb method not working correctly with 4 bit BMPs

martinberlin, 2020-09-04:
[image]
Any bitmap even if has complete white areas is rendering wrong with this function.
Can you please check why is rendering gray instead of the white in the image?
I tried this against another bitmap rendering methods like GxEPD and the white is coming white so there must be a bug in the way the bitmap is analysed.
In another topic would be great also to add PNG support since it's much more appealing than old windows bitmap format.

nitko12, 2020-09-04:
Hi, 
thank you for reaching out over your issue!

We are looking into PNG support and it might come soon.
And about BMPs not rendering correctly, it could be a color pallete issue. 
We will try to implement reading palletes as soon as possible and inform you of the fix once it's out!

martinberlin, 2020-09-04:
No problem thanks for researching. It's not a color palette issue. I can render the same bitmaps with my own code and the white is WHITE on the epaper. 
Cannot be a palette issue when I'm talking about 100% white.  Please try with this URL: 
http://img.cale.es/bmp/fasani/5ef94f52ad2f6

[DavidBowie.zip](https://github.com/e-radionicacom/Inkplate-6-Arduino-library/files/5173995/DavidBowie.zip)

MMore, 2020-09-05:
Hey @martinberlin, have you seen that the fifth parameter inverts the picture? Setting this parameter to true will flip all colors on the image, making black white and white black...

Just found that accidentally...

nitko12, 2020-09-07:
Hi @MMore, great suggestion!

This fixes this particular case, but it is true that we misinterpret BMPs slightly, particularly color tables...

Drawing monochrome bitmaps, as this Bowie one, should be done in monochome mode, obviously, but monochrome in bitmaps terms means color 0 or color 1 supplied in the color table. Problem is that there could be any 2 colors supplied, while out monochrome mode only uses black and white. Therefore when using monochromes you should test and invert the color for now, until we discuss this issue. 

Possible solutions are switching Inkplate into 3bit mode, or dithering original colors in monochrome mode. Or simply testing is color 0 actually darker than color indexed 1 and inverting if needed.

Fix will be issued soon, probably with a slight library rework.

Could you please provide files which didn't render correctly in your original issue, as it could be a different problem them being a 4 bit files.

:)

martinberlin, 2020-09-07:
Hi @nitko12,
Sure here you go: http://img.cale.es/bmp/fasani/5e5ff140694ee

I understand the monochrome / 4 bit thing, but that is something that as a user I should not worry about, the component should accept a monochrome and render it correctly independently of the bitdepth of the image. And that should be the norm. I don't have to invert the colors to get it right :)

closed 2020-09-07 by davidzovko

martinberlin, 2020-09-20:
> Hi,
> here is that image rendered using 4 bit palleted code from [here](https://github.com/nitko12/Inkplate-revision).
> 
> [image]
> 
> This confirms the issue is palette related which will be fixed, alongside few other bugs, once the library revision is done and tested.
> 
> Thanks for reporting this sneaky bug! :)

David can you tell me why I cannot reproduce this?

Using https://github.com/nitko12/Inkplate-revision.git: 
Rendering screen again from: http://img.cale.es/bmp/fasani/5ef94f52ad2f6
[E][WiFiClient.cpp:309] setOption(): fail on fd -1, errno: 9, "Bad file number"
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9

nitko12, 2020-09-21:
Could you try now,
 you can even try running test.ino in tests, I've left your image URL :)

Also we've moved revision branch to official repo here:

https://github.com/e-radionicacom/Inkplate-6-Arduino-library/tree/revision