SolderedElectronics/Inkplate-Arduino-library #33: Bitmap rendering does not respect DWORD row alignment

bstau, 2020-09-07:
When using the bitmap rendering functions such as `Inkplate::drawBitmapFromWeb()`, the renderer fails for bitmaps whose dimensions are not divisible by certain factors (for example, 8-bit images require a width divisible by 4, and 4bpp images require a width divisible by 8).

When using an uncompressed payload, the Win32/OS-2 bitmap format requires that pixels are required to be packed in rows that fit neatly into DWORDs. For example, an image of size `57x64` with 4 bits per pixel will have a `BITMAPINFOHEADER` struct of:

```
  .biSize=40,
  .biWidth = 57,
  .biHeight = 64,
  .biPlanes = 1,
  .biBitCount = 4,
  .biCompression = 0,
  .biImageSize = 0x800 // 2048; 32 bytes per line, not 29 as if byte packed
```

The error appears to be in the `Inkplate.cpp` library, specifically, in the `Inkplate::drawGrayscaleBitmap#Web()` collection of functions. Reads from `f_pointer` are not DWORD-aligned per row.

I'll prepare a pull request with the local changes I have in my copy of the library, but for those experiencing rendering issues with odd-sized bitmaps, you can work around these issues by padding (or cropping) your image to a multiple of 4 or 8, depending on the bit depth.

To calculate the line stride:

```
// Returns the line stride (in bytes) for an uncompressed Windows/OS-2 bitmap.
//
// In Windows-land, bitmap lines are rounded up to a DWORD (4 bytes), by padding
// the end of each line with unused data. The pixel data will not necessarily
// occupy the entire stride (especially in the case where (width % 8) > 0.
inline size_t BitmapLineStride(int8_t bpp, int width_pixels) {
  // Calculate the number of bits per line, round up to multiple of 32 by
  // truncation and bitmasking.
  return (((width_pixels * bpp) + 31) >> 3) & ~0x03;
}
```

`f_pointer` should be incremented by the row stride, not per-byte with individual reads from the bitmap when copying into `pixelBuffer`. For example:

```
        memcpy(pixelBuffer, f_pointer, w_bytes);
        f_pointer += w_bytes;
```

A number of [test cases](https://github.com/e-radionicacom/Inkplate-6-Arduino-library/files/5182163/test-cases-57x64-bitmaps.zip) have been attached that should be useful in verifying library behaviour.

nitko12, 2020-09-07:
Hi, 
we are working on a "rework" of many functionalities right now, 

this is the new bitmap drawing function, and I've linked you to the exact line calculating pixel row width [here](https://github.com/nitko12/Inkplate-revision/blob/5d94d15672524b17124b3ecefbd9ab89b4aacc61/src/include/ImageBMP.cpp#L107).

I hope this will be done soon, as it also fixes a few other bugs.

:)

bstau, 2020-09-07:
Awesome, looks like basically the same fix. Good stuff!

closed 2020-09-07 by bstau