SolderedElectronics/Inkplate-Arduino-library #35: Where did the weather go?

jf1452, 2020-09-12:
Hi,

I was looking forward to playing with the weather and calendar examples; but the links don't work; and there are no longer any arduino examples; did I look in the wrong place?

Cheers

Jon

davidzovko, 2020-09-12:
Hi Jon, you can find those in the "dev" branch. We are working on some changes to the library, so we had to remove those. It will be back to master branch soon!

jf1452, 2020-09-12:
Thank you so much David! very fast response. I'll take a look.

jf1452, 2020-09-12:
Hi David, I got it now; I see you're not far away :) I'm just over the border near Ivancna Gorica! When there are no restrictions I love going to Hrelic

davidzovko, 2020-09-13:
Well not that much close, but Hrelić is widely know :) - we are located in Osijek, at the east of Croatia. 

Let us know if you need any other kind of help!

closed 2020-09-13 by davidzovko