SolderedElectronics/Inkplate-Arduino-library #36: Determining battery voltage unreliable

mh-g, 2020-09-13:
I have a strong discrepancy between battery voltage measurements (using Display.readBattery()) depending on whether WiFi is on or not.

My program either (a) just updates the display with clock and internal measurements (which takes about 2 seconds) or (b) additionally activates WiFi, connects to an MQTT client, waits for several topics and then displays all the stuff (which takes about 4 seconds if all goes quickly).

In case (a) I get a measurement of about 3.55V, in case (b) of 3.80V (using a fully charged LiPo 1200mAh battery). Increasing the wait time to up to 10s does not change the measurements in either case.

davidzovko, 2020-09-15:
We think this is mostly due to the poor performance of ESP32 internal ADC. What is actual voltage on the battery, measured with a voltmeter, before WiFi and 10s after WiFi? 

My colleague will follow up with more info.

BornaBiro, 2020-09-15:
It would really help us if you can read battery voltage with some voltmeter/multimeter while WiFi is trying to connect to the network because there can be two problems;
- Battery can't handle high current demands while Inkplate is connecting to WiFi (especially if you are refreshing the screen at the same time) and voltage drops. You can try to use a more powerful battery or battery that has lower ESR.
- ADC of ESP32 is not accurate enough or it has big non-linearity. This is a known problem with the internal ESP32 ADC. It's has a very non-linear response for some reason. You can find some info about this issue on esp32 forum and on espressif github
(helpful links: https://github.com/espressif/esp-idf/issues/164, https://esp32.com/viewtopic.php?t=13089 )
You can try to compensate it somehow or you can try to calibrate it.

mh-g, 2020-09-17:
I did the measurements. Battery voltage is 4.17V according to a multimeter and does not change much during the trials (if needed, I can repeat using a scope, but this would have to wait till the weekend).

The relevant source code is at https://github.com/mh-g/InkplateDisplay/blob/master/InkplateDisplay.ino (lines 223-226 for the output). The difference between both cases is triggered by touching all three pads at the same time (line 592).

A display update without WiFi active yields ESP measured voltages of about 3.52V (65%).

A display update with WiFi active yields ESP measured voltages of 3.71V (88%).

As for the first link which suggests a 10% too low readout, this nicely fits (90% of 4.17V being 3.75V). But it still does not explain the values of 3.52V.

davidzovko, 2020-09-26:
Thanks for detailed info - that proves unexpected ESP32 ADC behaviour, but not so large variations. The measurement might have an offset or different value by some factor, but the offset/factor should stay the same all the time. 

To me, this seems like ESP32 ADC purely related issue. Did you manage to find any solution?

closed 2020-10-27 by davidzovko

mh-g, 2020-10-30:
I will most likely just implement a correction function.

davidzovko, 2020-11-05:
Fixed here: https://github.com/e-radionicacom/Inkplate-6-Arduino-library/commit/27c9b71f51b70f3b0df6750bcc0f46a550b408de

mh-g, 2020-11-16:
I am currently doing a longterm measurement series comparing the voltage between the pins in the battery socket and the result of display.readBattery(). The results are really weird: between about 3.9V to 4.5V of voltage measured at the pins, display.readBattery() gives about 1V less (2.9V to 3.6V). Then, readBattery() rapidly drops to 0.0V, while I can still measure about 3.8V at the pins. (See 
[image]
attachment.)

This is all before the fix 27c9b71. But I do not see how it would help.

So far, I do not see any negative effects on the lithium battery (like bulbing due to gas). But I haven't tried to charge it again ...

The good news is, if the measured battery voltage at the pins is correct, it will last really long. I just do not have any means to determine how long.

BornaBiro, 2020-11-16:
We did a little bit deeper investigation on this problem and it turns out: there was really a problem with battery measurement. MOSFET that turns on voltage divider for battery measurement wasn't turning on when you did battery measurement but it was on when you did not measure battery (inverted logic). So ADC did not measure anything useful (as you can see from display.readBattery() data). But not only that, voltage readings were wrong due to wrong ADC reference and attenuation settings. After the mentioned commit battery voltage measurement should work just fine.