SolderedElectronics/Inkplate-Arduino-library #37: reintroduce drawBitmapFromWeb

Gearme, 2020-09-13:
You appear to have deleted the drawBitmapFromWeb function again in the following commit. As martinberlin mentioned in a previous issue, it would be useful for (and is currently required by) his CALE client for the Inkplate 6. Any chance of getting it back on master? Or was this an intentional deletion?

https://github.com/e-radionicacom/Inkplate-6-Arduino-library/commit/c9c2031c66e106bab495a93e6137e25fd385ea45

davidzovko, 2020-09-13:
Hi, this is just temporarily. We are doing a library rework. We will return everything in a week or so. 

You can find the old library in the dev branch.

davidzovko, 2020-09-26:
Hello, the function is available using the _drawImage_ method, now in revision branch: https://github.com/e-radionicacom/Inkplate-6-Arduino-library/tree/revision

It will be released soon in the master branch and Arduino IDE library manager.

closed 2020-09-26 by davidzovko