SolderedElectronics/Inkplate-Arduino-library #38: rotation issue

jojo-Hub, 2020-09-26:
Hi,

I ran into an issue, using portrait mode. The example code below should draw a 100x100 px square, starting in the top left corner (which should also be (0,0)). Then it draws strings, printing their horizontal and vertical position. And there should be a top and a left rectangle, indented by 200px and spanning all over the display with 1px margin.

```c++
#include "Inkplate.h"
Inkplate display(INKPLATE_1BIT);

void setup() {
  display.begin();        //Init library (you should call this function ONLY ONCE)

  display.setRotation(1);
  // PROBLEMS:
  //   use rotation=1 (USB in top right corner): cannot draw on bottom ~200px
  //   use rotation=3 (USB in lower left corner): cannot draw on bottom ~200px, also need to set xshift to 200

  int xshift = 0; //200 for rotation==3
  int yshift = 0; //always 0 for initial alignment

  display.setCursor((display.width()+xshift)/2, (display.height()+yshift)/2); //print width & height ~center
  display.print(display.width());
  display.print(display.height()); 

  display.drawRect(1+xshift, 1+yshift, 100, 100, BLACK); // draw square in top left corner, 100x100 px, 1px margin to border
  display.drawRect(1+xshift, 200+yshift, 100, display.height()-yshift-200-1, BLACK); // draw left rect, from y=200 to margin-1px
  display.drawRect(200+xshift, 1+yshift, display.width()-xshift-200-1, 100, BLACK); // draw top rect, from x=200 to margin-1px

  for(int i=0;i<display.width(); i=i+20) { //draw horizontal index
    display.setCursor(i+xshift, 150+yshift);
    display.print(i);
  }

  for(int i=5;i<display.height(); i=i+10) { //draw vertical index
    display.setCursor(150+xshift, i+yshift);
    display.print(i);
  }

  display.display(); //draw.
}

void loop() {
}
```

However, for portrait mode (rotation == 1|3), I cannot write to the 200px at the bottom, the rectangle and vertical strings stop after about 400 px. Also for mode 3, I had to apply an x shift to the drawing in order to have (0,0) in the top left corner. As the top horizontal rectangle is "open" on the right (there is no vertical border drawn), I suspect an issue with the alignment/rotation of framebuffer vs. actual drawing.
The display width and height are always 800 x 600, regardless of the rotation parameter.

Regards,
Johannes

davidzovko, 2020-09-26:
Hello, thank you for reporting this error. 

We have implemented a fix to revision branch, the commit is here: https://github.com/e-radionicacom/Inkplate-6-Arduino-library/commit/b6781dd127b8119413297fecd2905f67881683c7

Can you please try now?

jojo-Hub, 2020-09-26:
Thanks! Now it works as ecpected

closed 2020-09-26 by jojo-Hub