SolderedElectronics/Inkplate-Arduino-library #39: Can compile examples

eLVas, 2020-09-29:
I can't compile any of the examples. When compiling I'm getting this error in ArduionoIDE:

```
Traceback (most recent call last):
  File "/home/elvas/.arduino15/packages/Croduino_Boards/tools/esptool_py/2.6.0/esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
ModuleNotFoundError: No module named 'serial'
```

Do I need to install anything else except `Croduino-Board-Definitions-for-Arduino-IDE`, `Inkplate-6-Arduino-library` and `Inkplate-6-SDFat-Arduino-Library`?

davidzovko, 2020-09-29:
Hello, did you make the first step from this tutorial: https://inkplate.io/getting-started/

_Install the Inkplate 6 board definition to add Inkplate 6 as a board in Arduino IDE: https://e-radionica.com/en/blog/add-inkplate-6-to-arduino-ide/

What is the board selected in your Tools->Board menu in Arduino IDE?

eLVas, 2020-09-29:
Hello @davidzovko, 
Yes I did install a bord definition and selected "Inkplate 6(ESP32)" in Boards
[image]

eLVas, 2020-09-29:
I have fixed it by following these [instructions](https://github.com/espressif/arduino-esp32/issues/13#issuecomment-252456782).

tldr: You need to install pyserial with `pip install pyserial`

closed 2020-09-29 by eLVas