SolderedElectronics/Inkplate-Arduino-library #44: Partial update only works in 1-bit mode

jf1452, 2020-10-27:
Inkplate display(INKPLATE_1BIT);
....
display.partialUpdate(); - Works

Inkplate display(INKPLATE_3BIT);
....
display.partialUpdate(); - Doesn't work

However, I was amazed to see how well PNG from Web works in 1 bit mode; I think the contrast is actually better than 3 bit mode.

nitko12, 2020-10-27:
Yes,
for now we don't have an implementation of 3 bit partialUpdate, primarily because of hardware limitations.
It would require larger amounts of memory to function correctly, 
but in the future we hope to add that as well :D

Anyways thanks for reaching out!

jf1452, 2020-10-27:
Thanks for the explanation nitko12, yes, that makes complete sense; after seeing how well dither works in 1 bit mode, I'm more than happy with the results.

closed 2020-10-27 by jf1452