SolderedElectronics/Inkplate-Arduino-library #47: Possible mistake in Inkplate.cpp

turgu1, 2020-11-22:
Hello all,

I found what I think to be a little mistake in the Inkplate::partialUpdate() function. The beginning of it:

```C++
void Inkplate::partialUpdate()
{
    if (getDisplayMode() == 1)
        return;
    if (_blockPartial == 1)
        display1b();

    uint16_t _pos = 59999;
```

should be:

```C++
void Inkplate::partialUpdate()
{
    if (getDisplayMode() == 1)
        return;
    if (_blockPartial == 1) {
        display1b();
        return;
    }

    uint16_t _pos = 59999;
```

There is no need to run the rest of the partialUpdate() function when the code is forcing (_blockPartial == 1) the use of the display1b() function. It doesn't hurt the end result but is not useful.

Cheers!!!
Guy

BornaBiro, 2020-11-25:
I really don't know how we missed return after 1 bit mode forced refresh, but return definitely should be after full screen update and it should not write to screen anything else after.
If this is done repeatedly it could cause burn in.
We will fix this ASAP!

Thanks for the report!

turgu1, 2020-11-25:
Hello Borna,
Can we exchange some e-mail off github. I've some questions for you in regard of the refresh algorithm.
Thanks
Guy: turgu666@gmail.com

BornaBiro, 2020-11-26:
No problem.
I send you an e-mail. :)

BornaBiro, 2020-11-26:
Bug is fixed in https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/067fb7c77c928ba837d1e7956a83bb4c6565dfe6

closed 2020-11-26 by davidzovko