SolderedElectronics/Inkplate-Arduino-library #48: Questions regarding the API

bbernhard, 2020-11-23:
Hi, 

I would have some questions regarding the API: 

* In the `Image.h` file some lower level methods are marked as deprecated (e.g `drawJpegFromSd`, `drawJpegFromWeb`) and it is advised to use `drawImage` instead. Unfortunately, I can't use `drawImage`, as the method directly checks for the file ending to determine which low level function to call. But I would like to draw images from URLs that use UUIDs instead of file names. So e.g my URL looks like this: `http://127.0.0.1:8080/img/57913c70-ff5d-4a5d-84f3-d49ec79b5a9c `

Would it be possible to add another member function overload which lets you specify the image type?
So e.g something like that: `bool drawImage(const char *path, int x, int y, const bool ImageFormat& imageFormat, bool dither = 1, bool invert = 0);`

* Something that I would find quite useful is, the possibility to center an image on the screen. I saw that there's already a function called `getJpgSize` to get the image size for jpg images. So I quickly hacked together a `drawJpegFromWebCentered` helper function to test it and it seems to be working. 

Are there any plans to add such a functionality in the near future?

If you don't have time, but you accept pull requests I could open a PR for those two extensions.

nitko12, 2020-11-24:
Hi,
we'd be happy to see you contributing by making a PR.
But how would you specify image type exactly? Maybe declare an enum?

Also if you'd be making a centered image method, you can make a PR with your function, and we will add support for other image types, such as png and bmp. 

Just be sure to PR to revision branch.

nitko12, 2020-11-24:
Closed due to a PR fixing the issue.

https://github.com/e-radionicacom/Inkplate-Arduino-library/pull/49

closed 2020-11-24 by nitko12