SolderedElectronics/Inkplate-Arduino-library #52: Impossible to compile an example (library install issue)

Trytodosth, 2020-12-12:
Hi,

I use my Inkplate on my PC for several monthes now.
I am trying to setup Arduino on another PC. I am facing issues, even to run an example.

I have the feeling you updated the Board Manager and included the Inkplate 10. It don't remember it being like that as a separate menu item.
I tried 4 times, even rebooting between steps and trying to understand.

I really think I did things correctly, but maybe not. I can try more if not reproducable!

## Steps:
1. Windows 10, 64 bits
2. CH340 drivers installed following https://e-radionica.com/en/blog/ch340-driver-installation-croduino-basic3-nova2/ 
3. Fresh install Arduino IDE, having the `%APPDATA%/local` and `My Documents/Arduino` folders removed
4. Adding Inkplate board following https://e-radionica.com/en/blog/add-inkplate-6-to-arduino-ide/
[image]
It includes the github library ?? : "%AppData%\Local\Arduino15\packages\Croduino_Boards\hardware\Inkplate\1.0.0\libraries\Inkplate-6-Arduino-library-master".
5. Downloading the latest .zip from this repo. Renaming it as `InkplateLibrary.zip`.
6. Adding .ZIP Library. Message library added
7. Load the example sketch: `Inkplate_basic_BW`. Try to compile.
Result:
[image]

## Details of the error
```
Arduino: 1.8.13 (Windows 10), Board: "Inkplate 6 (ESP32), Enabled, Huge APP (3MB No OTA), 240MHz (WiFi/BT), 115200, None"

WARNING: library SPI claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on Inkplate architecture(s).

WARNING: library HTTPClient claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on Inkplate architecture(s).

WARNING: library WiFiClientSecure claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on Inkplate architecture(s).

WARNING: library Wire claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on Inkplate architecture(s).

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp: In member function 'bool NetworkClient::joinAP(const char*, const char*)':

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:21:10: error: 'class WiFiClass' has no member named 'mode'

     WiFi.mode(WIFI_MODE_STA);

          ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:21:15: error: 'WIFI_MODE_STA' was not declared in this scope

     WiFi.mode(WIFI_MODE_STA);

               ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:22:26: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

     WiFi.begin(ssid, pass);

                          ^

In file included from C:\Users\mhjap\AppData\Local\Arduino15\packages\Croduino_Boards\hardware\Inkplate\1.0.0\libraries\WiFiClientSecure\src/WiFiClientSecure.h:25:0,

                 from C:\Users\mhjap\AppData\Local\Arduino15\packages\Croduino_Boards\hardware\Inkplate\1.0.0\libraries\HTTPClient\src/HTTPClient.h:35,

                 from C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.h:21,

                 from C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:17:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:79:9: note:   initializing argument 1 of 'int WiFiClass::begin(char*, const char*)'

     int begin(char* ssid, const char *passphrase);

         ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp: In member function 'void NetworkClient::disconnect()':

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:41:10: error: 'class WiFiClass' has no member named 'mode'

     WiFi.mode(WIFI_OFF);

          ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:41:15: error: 'WIFI_OFF' was not declared in this scope

     WiFi.mode(WIFI_OFF);

               ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp: In member function 'uint8_t* NetworkClient::downloadFile(const char*, int32_t*)':

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:55:23: error: 'class WiFiClass' has no member named 'getSleep'

     bool sleep = WiFi.getSleep();

                       ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:56:10: error: 'class WiFiClass' has no member named 'setSleep'

     WiFi.setSleep(false);

          ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:59:22: error: 'class WiFiClient' has no member named 'setNoDelay'

     http.getStream().setNoDelay(true);

                      ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:98:10: error: 'class WiFiClass' has no member named 'setSleep'

     WiFi.setSleep(sleep);

          ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp: In member function 'uint8_t* NetworkClient::downloadFile(WiFiClient*, int32_t)':

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:108:23: error: 'class WiFiClass' has no member named 'getSleep'

     bool sleep = WiFi.getSleep();

                       ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:109:10: error: 'class WiFiClass' has no member named 'setSleep'

     WiFi.setSleep(false);

          ^

C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master\src\include\NetworkClient.cpp:131:10: error: 'class WiFiClass' has no member named 'setSleep'

     WiFi.setSleep(sleep);

          ^

Multiple libraries were found for "Inkplate.h"

 Used: C:\Users\mhjap\OneDrive\Documents\Arduino\libraries\Inkplate-Arduino-library-master

 Not used: C:\Users\mhjap\AppData\Local\Arduino15\packages\Croduino_Boards\hardware\Inkplate\1.0.0\libraries\Inkplate-6-Arduino-library-master

Multiple libraries were found for "WiFiClient.h"

 Used: C:\Program Files (x86)\Arduino\libraries\WiFi

 Not used: C:\Users\mhjap\AppData\Local\Arduino15\packages\Croduino_Boards\hardware\Inkplate\1.0.0\libraries\WiFi

exit status 1

Error compiling for board Inkplate 6 (ESP32).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

nitko12, 2020-12-14:
Hi,

thank you for reporting an issue.

We are working on a fix, possibly with a new board definition and
will send you a fix once it's done, to test on your system :)

nitko12, 2020-12-16:
Could you try updatind your board definition and uploading now?

There was an issue where it used an old library that was shipped with the board definition.

There also a known issue with pyserial in the board definition, but that shouldn't have an issue in your case.

Trytodosth, 2020-12-16:
Uninstalling and reinstalling the board definition worked.
Thanks!

closed 2020-12-16 by Trytodosth