SolderedElectronics/Inkplate-Arduino-library #53: Draw PNG from web appears to be broken with new library

jf1452, 2020-12-14:
Using OpenWeather_station_example.

Previous library worked fine; I have updated the library based on the new update; but this seems to have broken:
    sprintf(Output, "http://openweathermap.org/img/wn/%s@2x.png", OWOC.forecast[day].icon);
    display.drawImage(Output, dayPos + 10, dayOffset, true, true);

The new library draws the first PNG, but no further images.

After restoring to the previous version; all PNGs are retrieved and drawn from the web.

nitko12, 2020-12-14:
Hi,

could you please specify which commit worked correctly, and which one causes trouble? :)

jf1452, 2020-12-14:
Just checked; the one which worked is 14 commits behind the latest master.
https://github.com/jf1452/Inkplate-6-Arduino-library

nitko12, 2020-12-16:
I've tried drawing PNG's from current master and everything works as expected, could you send us the exact file not being rendered?

It's drawing 10 images in a row, without any problems.

Also could you please check if it renders other png's on your board and library?
If not, how about jpegs or other formats, if they're rendered correctly then the library we're using to render PNG's might be causing touble?

jf1452, 2020-12-16:
The code I wrote for OpenWeather only has access to png icons for all the images; have you tried my open weather example with the latest libraries? works fine with the version I pushed, but doesn't work with the latest version.

nitko12, 2020-12-17:
Hi,

there were a few bugs in the library, probably memory related.

I've done a quick fix here:
https://github.com/e-radionicacom/Inkplate-Arduino-library/tree/inkplate10-integration/examples/Projects/OpenWeather_station_example

It will be pushed to master, along other fixes, with this branch once it's finished and tested.

In the future, try to minimize dynamic allocation, as the memory managment system on the esp is pretty primitive :)

jf1452, 2020-12-17:
Many thanks @nitko12 ; your fix has got it working, looking good!

closed 2020-12-17 by jf1452