SolderedElectronics/Inkplate-Arduino-library #54: Draw BMP and JPG not as expected after update

martinberlin, 2020-12-17:
Hello, after getting this email alert that was an update in the Arduino library, I just deleted the library from Platformio **.pio/lbdeps** folder so it loads the new one.
I tried to output from my web-service [CALE.es](https://cale.es) 2 different formats, this is just a Grayscale black & white photos gallery:

1. BMP 4 bits: Partial success. Some images render, some not, usually black is never 100% black.
    http://img.cale.es/bmp/fasani/5ef94f52ad2f6
2. JPEG 24 bits: Does not work for me. Just refreshes and stays blank.
   JPEG 4 bits: Same as BMP 4 bits. 
    http://img.cale.es/jpg/fasani/5ea1dec401890

Other than that I don't see anymore the display.partialUpdate(); rendered like before. It just renders a completely empty screen. 

Before I've seen the "Connecting to WiFi..." text printed with partial update.
The sketch I'm using is: https://github.com/martinberlin/inkplate-platformio/blob/master/examples/CALE/cales-slides.cpp

Following the examples. Did I miss something?
I really want to see the new library rendering a black and white photo with grayscales.

nitko12, 2020-12-17:
Hi,

partialUpdate never worked in 3 bit mode?

You can switch modes in code with selectDisplayMode.
See here: https://inkplate.readthedocs.io/en/latest/arduino.html#inkplate-selectdisplaymode

As for the images not rendering, I'll test that out right now, as we got another issue that might be related
https://github.com/e-radionicacom/Inkplate-Arduino-library/issues/53

martinberlin, 2020-12-17:
Great then I cannot just render images in 3BIT mode and use partial update at the same time?
Good sorry didn't see that 53 was related since it's using PNGs. So if that's the same you can close this and just keep one of the issues

nitko12, 2020-12-17:
Yep,

3bit mode can't have partial updates (for now, we're working on it :)

Issue 53 actually wasn't related, it was probably a memory issue.

As for the images not rendering in 1 bit, are you using drawImage?
It's desireable to have file extension in the link/file, otherwise we cannot find image format that easily.

You can use old functions that draw a specific type for now, but when the Inkplate 10 integration is pushed to master, there will be an override like as suggested here:
https://github.com/e-radionicacom/Inkplate-Arduino-library/issues/48

If you're feeling lucky, you can try it out in the other branch, though it's still work in progress and will probably be fully functional in a few days :)

davidzovko, 2020-12-17:
Correct, you can not use 3-bit + partial update. It's super tricky to implement, not even Kindle has it.

martinberlin, 2020-12-17:
> As for the images not rendering in 1 bit, are you using drawImage?
It's desireable to have file extension in the link/file, otherwise we cannot find image format that easily.

Will try that. As per image type, can’t you just add a setter in the class?
Then I can just add it as a parameter and say drawImage(extra param **type** jpg)

nitko12, 2020-12-17:
Yes, 

that will be added soon, probably in the next major release :)

martinberlin, 2020-12-19:
Good but I really don’t see the point in closing this @nitko12 and @davidzovko when I still didn’t had time to try it. And so far I could not render a decent picture with grayscale in the Inkplate all the times I tried.

davidzovko, 2020-12-19:
Fair enough, I have reopened it. Can you please try adding the extension (JPG/BMP/PNG) to the filename and try rendering the image?

martinberlin, 2020-12-20:
Sure this afternoon I will save some of this images in an SD with proper extension and try to render them so I can write some more details here

martinberlin, 2020-12-21:
Found time to test it and make some pictures with the iPad. 
At the beginning I explored the drawBitmapFromWeb method. I assume because of the name that it only supports BMP format.
This is not working good, even with the proper extension (Actually adding a extension does not modify anything)

This image that looks quite bad with this method, looks good using drawImage from SD: 
[image]
[Original image preview](https://raw.githubusercontent.com/martinberlin/inkplate-platformio/master/examples/2.%20Advanced%20Inkplate%20Features/5-Inkplate_SD_pictures/4-bit-1.bmp)

Used this example: 
https://github.com/martinberlin/inkplate-platformio/tree/master/examples/2.%20Advanced%20Inkplate%20Features/5-Inkplate_SD_pictures

Second example looks quite better:
[image]
[Original image preview](https://raw.githubusercontent.com/martinberlin/inkplate-platformio/master/examples/2.%20Advanced%20Inkplate%20Features/5-Inkplate_SD_pictures/4-bit-2.bmp)

In resume, drawBitmapFromWeb() should be either refactored, or an example offered where you just download the image with one method and with the next one you send it to drawImage.
Using the extension to guess the image type is not a wise idea since you can use an image service that delivers no extension. Apart of that exactly for this reason mimeTypes exist, and it can be guessed analyzing the response header, but just because the server can send the wrong header in my opinion the best would be to add the type to drawImage as a parameter and then it will be easier to use. 
Left a JPG pool of images without extension here:
https://cale.es/backend/screen/render/5ea1dec401890

nitko12, 2020-12-22:
Hi,

as I said we will be adding (already in inkplate10-integration branch)
drawImage override that supports exact image type.

e.g.
display.drawImage("url", display.JPG, 0, 0);

As for these, it seems to me like a palette issue.

Could you send me the exact files.
I've tried accessing your link but it gives me a 404.

martinberlin, 2020-12-22:
Sure here the files I tried:
http://slosarek.eu/e/1.jpg
http://slosarek.eu/e/2.jpg
http://slosarek.eu/e/4-bit-1.bmp
http://slosarek.eu/e/4-bit-2.bmp
http://slosarek.eu/e/4-bit-3.bmp

nitko12, 2020-12-22:
Could you try using Inkplate10 integration branch?

It renders everything fine for me, except jpegs whose type format is not supported, probably some compression method not supported by the library.
You could try reencoding them, as we can't fix this issue. We rely on this https://github.com/Bodmer/TJpg_Decoder.

martinberlin, 2020-12-22:
I will try in the next days. But I do not understand what is wrong with the JPG. There is not any special compression, there are just normal Jpegs and they open with everything I tried.

nitko12, 2020-12-23:
You can find more info on Bodmers repo.

As stated there:

> Jpeg files in the "Progressive" format (where image data is compressed in multiple passes with progressively higher detail) are not supported since this would require much more memory, or too many Inverse Discrete Cosine Transform's for typical embedded systems.

These jpegs render normal in any modern viewer, but because the esp has very low performance and memory it can't handle them.
It could be that your jpegs are like that.

 Try rendering some other files, they should work correctly.

martinberlin, 2020-12-23:
Ok I will check more. You can close this now if bothers to have it open.
At least we know what it is. Will try the Inkplate 10 lib when I find some time. It would be great if there is a drawImageFromWeb where a type parameter could be passed. This can be just a wrapper function that internally calls drawImage

davidzovko, 2020-12-28:
Please let us know when you have checked, I'll leave the issue open, so we can know the exact issue source.

martinberlin, 2021-01-03:
Sure sorry I was on holidays but tomorrow I return to my duties so I will make myself some time to try it out. 
My idea to port this to IDF is also there but was delayed since I had too much work and little space in my brain for anything else. But I'm receiving in some weeks an epaper from LylyGO that is also parallel so I will try to find my way to do it. 
One question, sorry to get out of Issue main thema, is possible still to do partial updates switching from GRAY mode to the other mode just for this update? Or this is supposed to be called just on initialization of the epaper.

nitko12, 2021-01-07:
Hi,

the buffer is cleared when switching display modes, so it wouldn't draw anything you had before.

You can still draw the same things you did in grayscale to 1bit mode once you switched tho.

martinberlin, 2021-01-23:
I will be checking next week this. I must say that also using the drawBitmapFromWeb like in [this example](https://github.com/martinberlin/inkplate-platformio/blob/master/examples/CALE/cale-slides.cpp) is giving me trouble and killing my users battery. To download a single Bitmap is making at least 3 requests as I can see in the logs. And only on the last one is downloading the image. 
There must be something wrong. This does not happens with my Firmware in any version (Arduino-esp32 or ESP-IDF)

nitko12, 2021-02-11:
Are you sure it's not due to your internet connection or something?

Looking at the verbose output of esp32 I get this:

[image]

Could you supply more info :)

martinberlin, 2021-02-11:
Sure, this is what I see in CALE.es database logs: 

[image]

As you see only on the last request it finally load the image. This is what my user Andrew told me when I asked:

>
I set it to wake and request an image every 30min. 

However it will retry up to 3 times if there’s any error. It’s an Inkplate 6 and for some reason their code to get an image from URL always fails the first time with an error message while the second attempt usually succeeds. 

Don’t know if you noticed similar behavior on Inkplate 6? I’ll publish the code I’m using to my GitHub later today, needs a bit of cleanup. 

I based it on your code @ 
https://github.com/martinberlin/inkplate-platformio/tree/master/examples/CALE and only added deep sleep for 30min. 

Maybe there is some mistake in my code? Maybe he just put a too short delay, but I don't think thatÄs the problem, since I also experience that it tries to load 2 or 3 times and only on the last try it loads.

ndoo, 2021-02-13:
@nitko12 this is the code I am using:
https://github.com/ndoo/inkplate-platformio/commit/9f9c59e42a3e92a2e8a87a32de4ef8cac94fc574

nitko12, 2021-02-16:
Hi,

so after further testing, I can't seem to find what's causing your bug :( 

As I've said, serial monitor logs all check out, and I've tested creating a small web server to see if I'd get the same result there with your code, but I get only one request:

[image]

Here is my server code https://repl.it/@nitko12/MeaslyTidyLinux#index.js, you can fork it and test it yourself.

Could the issue be with your server setup? Like content lenght header or something?

martinberlin, 2021-02-16:
>Could the issue be with your server setup? Like content lenght header or something?

If that would be I think it will cause the same problem with my requests. Can you try updating the image URL with one of my delivery service?

http://img.cale.es/bmp/fasani/5ef94f52ad2f6

nitko12, 2021-02-17:
Sure, I've uploaded your code to an Inkplate and will leave it to refresh for some time.

You can check the logs if they're the same as with your requests.

In the esp's serial it all seems fine, one request to download and that's it.

martinberlin, 2021-02-18:
No problem I will check and report to you later.

martinberlin, 2021-02-18:
For me what is buggy is drawBitmapFromWeb. It gives this error (And it's not the .bmp missing extension)

[E][WiFiClient.cpp:309] setOption(): fail on fd -1, errno: 9, "Bad file number"
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9

So I need to do this in order to load an image: 

```c++
  if (!display.drawBitmapFromWeb(screenUrl, 0, 110, false)) {
    if (!display.drawBitmapFromWeb(screenUrl, 0, 110, false)) {
      display.drawBitmapFromWeb(screenUrl, 0, 110, false);
    }
  }
```

nitko12, 2021-02-19:
That doesn't seem to be an error, I get the same when loading other bmp's from the web.

[image]

Also, yes, I've noticed that your link seems to work only every third time, 
but that seems to be due to your site, as other links are opened on the first try.

Could you also try other sites please?

martinberlin, 2021-02-19:
>but that seems to be due to your site, as other links are opened on the first try.

But can you explain why?
I don't have any problems loading this both with my Arduino-esp32 version of the [Firmware](https://github.com/martinberlin/eink-calendar) neither with Cale-idf. They both open at the very first time.
But in this Firmware, it seems like it's a random bug, it sometimes open at the first time. Sometimes at the second call or the third. 

It will be too easy to rule out that the problem is on the site. But it cannot be when I can download images on the first time always with my own Firmware. Specially when other 80 users are doing that without any problem so far (That is except when they use Radionica Firmware)

davidzovko, 2021-02-19:
We do not have an answer right now. All tests we have made do not show this behaviour, requests are successful on the first try, except for the CALE. We'll do some more tests.

martinberlin, 2021-02-19:
No problem. I hope it will be solved, it will be really nice not to have this random errors, I can help debugging. Just checked the headers of a BMP request: 
http://img.cale.es/bmp/fasani/5ef94f52ad2f6

```
HTTP/1.1 200 OK
Date: Fri, 19 Feb 2021 08:48:16 GMT
Server: Apache/2.4.46 (Amazon) OpenSSL/1.0.2k-fips PHP/7.2.34
X-Powered-By: PHP/7.2.34
Cache-Control: no-cache, private, max-age=172800
Content-Length: 60062
Expires: Sun, 21 Feb 2021 08:48:16 GMT
Connection: keep-alive, Keep-Alive
Keep-Alive: timeout=5, max=100
Content-Type: image/bmp
```

They seem fine to me. I think there must be something in the request itself that is rejecting this. I cannot imagine that is something about how the server delivers the bitmap.

nitko12, 2021-02-22:
Hmm,

did you check out my code, is there anything suspicious?

Headers do seem fine,
these are the ones from the test server which work fine every time:
[image]

martinberlin, 2021-02-22:
Wait but what code are you referring to?
My point is: This is working right on both Arduino-esp32 ([eink-calendar Repository](https://github.com/martinberlin/eink-calendar)) and also in Cale-idf (With both about 40 users)

Why is doing this random error in your repository?
I tried myself and I still could not discover why. But for sure the image delivery is not the issue since I can download the image from another Firmwares in **one try always** without any problem.

nitko12, 2021-02-24:
I see,

I was talking about our code for downloading images here:
https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/c4149543708861b96b5a47f4ae4e728628eb0a16/src/include/NetworkClient.cpp#L49

as I understand your code downloads and processes the image at the same time, while we first download it into ram and then process it and display.

Not sure if any of this can result in a bug, I presume not, do you have any ideas?

martinberlin, 2021-02-25:
Yeah is true. I wanted to make in "Streaming mode" since with BMP that is possible (I guess not with PNG or JPEG)
Mainly since I wanted to support ESP32 without PSRAM. I will try to review later this downloadFile and see if I see something wrong. A friend is using now my Inkplate 6 at his office so I will test next time I visit.
I could not see any more logs that can shed more light on this but for sure I will post the whole thing next time I try.

nitko12, 2021-08-10:
Feel free to reopen if you have done any progress, 

I'll close this now as the issue appears stale :)

closed 2021-08-10 by nitko12