SolderedElectronics/Inkplate-Arduino-library #56: Unexpected behavior under battery only

bwhitman, 2020-12-28:
Inkplate 6. With a 2000 3.7V battery connected but not anything in the USB port, the system does not act properly. In this sketch, the screen will flash every 5 seconds (as if it is rebooting; i cannot see any debug info with no USB cable connected) but not show any picture. With a USB cable connected to the PROG port, the sketch works fine and shows a picture from the internet. I've charged this battery both with the Inkplate charger, and to test, another known good charger until it was full. Tested the battery in another non-Inkplate ESP32 device to confirm it's fine.

`
#include "HTTPClient.h"          //Include library for HTTPClient
#include "Inkplate.h"            //Include Inkplate library to the sketch
#include "WiFi.h"                //Include library for WiFi
Inkplate display(INKPLATE_3BIT); 
#include "driver/rtc_io.h" // ESP32 library used for deep sleep and RTC wake up pins
#define uS_TO_S_FACTOR 1000000L // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  1800L      // How long ESP32 will be in deep sleep (in seconds)
RTC_DATA_ATTR int slide = 0;

const char *ssid = "SSID";     // Your WiFi SSID
const char *password = "PASSWORD"; // Your WiFi password

void setup()
{
    // Connect to the WiFi network.
    WiFi.mode(WIFI_MODE_STA);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
    }
    
    display.begin();        // Init Inkplate library (you should call this function ONLY ONCE)
    display.clearDisplay(); // Clear frame buffer of display
    display.display();
    
    display.drawImage("https://URL/image.png", 0, 0, false, false);
    display.display();

    delay(5000);
    WiFi.mode(WIFI_OFF);
    rtc_gpio_isolate(GPIO_NUM_12); // Isolate/disable GPIO12 on ESP32 (only to reduce power consumption in sleep)
    esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR); // Activate wake-up timer 
    esp_deep_sleep_start();                                        // Put ESP32 into deep sleep. Program stops here.
}

void loop()
{
    // Nothing...
}`

davidzovko, 2020-12-28:
Hey, sorry, that's a hardware issue. It's fixed on newer production runs, but you have an early one. It is related to the capacitor above the reset button: https://postlmg.cc/rdQG2SSN

I tried to sketch it there on the image, if you are able to, you should add a capacitor in parallel with the existing one. It should be at least 2.2uF.  Just any of that value of higher should be okay. 

Let me know if you can try that

bwhitman, 2020-12-30:
Ah, too bad I got a bad model. I reached out to the team over Crowdsupply for a fixed replacement but haven't heard back yet. I tried to add that capacitor but it did not work for me.

davidzovko, 2020-12-30:
Yeah, sorry about it. We will handle the replacement for you. Please contact us at kontakt@e-radionica.com.

I will close the issue now.

closed 2020-12-30 by davidzovko

jakobwesthoff, 2021-01-16:
It seems to me I have got mostly the same problem with my inkplate. I never really got time since I received it. However know playing around with it, I got a problem as soon as a battery is attached. If no USB is attached at the same time the inkplate does not seem to stay in sleep mode. I updates the screen quite fine however. Unfortunately it then seems to wakeup immediately again, without sleeping. Powered by USB everything works fine however. Could this be the same problem? Is there a way to identify this "older" hw revision with this problem?

davidzovko, 2021-01-21:
Yep, definitely sounds like exactly the same problem. Unfortunately, no way to recognise the old version because they are using the same PCB. But based on your description, it's a safe bet that adding a capacitor as described above will work: https://postlmg.cc/rdQG2SSN

Can you try that?

jakobwesthoff, 2021-01-21:
I already contacted you via email about this. We already agreed on a replacement delivery. Both packages are already in transit. Thanks for the quick answer.

davidzovko, 2021-01-21:
Well, they are fast. Good to hear this is resolved.

indefini, 2021-02-05:
@davidzovko 
I had the same problem and I replaced the capacitor with an 2.2uf one and it fixed it (it boots on the battery without blinking)
But I have another problem and I am not sure if it is related or not.
On usb or battery, the inkplate stops updating after a few minutes... sometimes it 5mins, sometimes 20-30minutes max
For example I upload the weather example and after 4mins the clock stops updating.

davidzovko, 2021-02-05:
Hey, that sounds like a software issue to me. Are you using the unchanged code from the repo? What are your Arduino IDE settings?

jkmaxwell, 2021-02-07:
I'm uncertain but I may have this problem too. Can you please verify:

1. I have connected a 503759 battery to the battery socket
2. When on USB power, the unit works fine
3. When I disconnect, the unit does not power cycle or show any activity based on the power switch

My assumption was that
- the battery would charge while plugged into USB power
- toggling the power switch should show a redraw or screen flash to verify something is happening

I just bought mine last week via eradionica shipped to the US

indefini, 2021-02-07:
@davidzovko the code is unchanged (apart from wifi settings) from the library downloaded from the arduino ide, version 1.1.0.
I think settings are default : 
upload speed : 115200
CPU frequency : 240Mhz
Partition Scheme : Huge App
Core Debug : None
PSRAM : enabled

davidzovko, 2021-02-09:
@indefini the settings look good. Does this happen with any code, or just weather example?

@jkmaxwell did putting the battery polarity correctly fix the issue?

jkmaxwell, 2021-02-09:
@davidzovko yes. thanks.

indefini, 2021-02-16:
@davidzovko  I think examples that don't use wifi looks fine? like the game of life or Inkplate_Low_power looks fine.
the weather example (daily and hourly) ends up with this error : 
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x22 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download
```
which seems like https://github.com/espressif/arduino-esp32/issues/577

weather station seems to work but sometimes reboot
```
E (163717) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (163717) task_wdt:  - IDLE0 (CPU 0)
E (163717) task_wdt: Tasks currently running:
E (163717) task_wdt: CPU 0: wifi
E (163717) task_wdt: CPU 1: loopTask
E (163717) task_wdt: Aborting.
abort() was called at PC 0x400eee43 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008b51c:0x3ffbe360 0x4008b799:0x3ffbe380 0x400eee43:0x3ffbe3a0 0x40082261:0x3ffbe3c0 0x40162086:0x3ffd5df0 0x40155327:0x3ffd5e10 0x4014d22c:0x3ffd5e30 0x4013dc37:0x3ffd5e50 0x4014ff63:0x3ffd5eb0 0x4014aa29:0x3ffd5ed0 0x4014fa49:0x3ffd5f40 0x4014cda4:0x3ffd5f60 0x4014cebe:0x3ffd5f80 0x4015c3d2:0x3ffd5fa0 0x4008c062:0x3ffd5fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10900
load:0x40080400,len:6360
entry 0x400806b4
Serial Monitor Initialized
Welcome to Wol Inkplate 6 weather example!

```

the crypto_tracker example also seems fine but also reboots
the serial monitor has several output like : 
```
E (29030) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (29030) task_wdt:  - IDLE0 (CPU 0)
E (29030) task_wdt: Tasks currently running:
E (29030) task_wdt: CPU 0: wifi
E (29030) task_wdt: CPU 1: IDLE1
E (29030) task_wdt: Aborting.
abort() was called at PC 0x400e88db on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008b540:0x3ffbe4a0 0x4008b7bd:0x3ffbe4c0 0x400e88db:0x3ffbe4e0 0x40082265:0x3ffbe500 0x4015df80:0x3ffdc370 0x4015121b:0x3ffdc390 0x401490cc:0x3ffdc3b0 0x40139aef:0x3ffdc3d0 0x4014be0f:0x3ffdc430 0x401468c5:0x3ffdc450 0x4014b8f1:0x3ffdc4c0 0x40148c44:0x3ffdc4e0 0x40148d5e:0x3ffdc500 0x401582ce:0x3ffdc520 0x4008c086:0x3ffdc550

Rebooting...
ets Jun  8 2016 00:22:57
```

hoytech, 2021-03-14:
I did this mod and it seemed to fix it for me:

[image]

I also had to super-glue the screen to the PCB since the tape was eroded or something, and it almost dropped and broke the first time I picked it up.

Still, I love this device!

smithzvk, 2021-07-07:
I apologize for resurrecting a dead thread, but is there a chance that putting too much capacitance here can prevent flashing via USB?  I added ~8uF in parallel and it seems to fix the repetitive resets (just the first SMT that I could locate which fit and was > 2.2 uF), but I can no longer program the board.  Unfortunately, I also upgraded my OS (PopOS 20.04 to 21.04) between the now and the last successful flash, so I can't rule out some software change.  esp-tool fails with:

```
arduino-cli upload -p /dev/ttyUSB0 --fqbn Croduino_Boards:Inkplate:Inkplate6 ./
esptool.py v3.0-dev
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
Error during Upload: uploading error: uploading error: exit status 2
```

Serial log shows the following repeating with each attempt to start the flash:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10900
load:0x40080400,len:6360
entry 0x400806b4
```

Looks like the attempted esp-tool command is:

```
python ".../packages/Croduino_Boards/tools/esptool_py/3.0.0/esptool.py" --chip esp32 --port "/dev/ttyUSB0" --baud 115200  --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0xe000 ".../packages/Croduino_Boards/hardware/Inkplate/1.0.1/tools/partitions/boot_app0.bin" 0x1000 ".../packages/Croduino_Boards/hardware/Inkplate/1.0.1/tools/sdk/bin/bootloader_dio_40m.bin" 0x10000 "/tmp/arduino-sketch-C06304407241DEE7491921B8783502FB/ebook_reader.ino.bin" 0x8000 "/tmp/arduino-sketch-C06304407241DEE7491921B8783502FB/ebook_reader.ino.partitions.bin"
```

Any help is appreciated.  Thanks.

davidzovko, 2021-07-07:
Most likely the capacitor is too big, as it looks like that board can not reset. Please try reducing it to 2-ish uF.

smithzvk, 2021-07-08:
Thanks.  Still couldn't find a ~2uF, but wanted something that matched the board and found one measuring 4.2 uF and stuck in on there.  All seems to be working now.

mattncsu, 2021-08-31:
2.2uF in parallel worked for me too.  FYI @hoytech, if you use a small knife on the backside of the battery connector, you can pull the contacts out so you can swap them without having to cut and resolder.

hoytech, 2021-08-31:
@mattncsu - Thanks for the tip, I'll give it a try!

Zylanx, 2021-11-30:
Did this, cap ended up lifting the pad after it got knocked slightly
Well... dunno what to do now

theswampire, 2022-01-18:
Hi there. I too have just stumbled upon this problem 👀.
So before I scavenge the rest of my old devices for capacitors, I wondered if I have to consider the specific voltage rating of a  capacitor.
If yes, what would it be?

davidzovko, 2022-01-19:
Hey, just any voltage above 6V will work. 

@Zylanx sorry about that! Do you have a photo, maybe it can be fixed?