SolderedElectronics/Inkplate-Arduino-library #57: PINs documentation: Would like to try EPDiy component (ESP-IDF)

martinberlin, 2021-01-22:
Hey I know this is not strictly categorized as Bug or Issue but I would like to see if it's possible to try this EPDiy component since it works great and that way I can add also support differently in my CALE Firmware.
In their [supported displays list](https://github.com/vroland/epdiy#displays) the Inkplate 6 is there so it should work. 

But what I still don't get is how are the Control lines done in the Inkplate. 

Inkplate with EPDiy
PINS -> What are this ones?
#define WAKEUP 3
#define PWRUP 4
#define VCOM 5

CKV 01
SPH 02
LE   04  // Latch enable
DATA PINS  v Inkplate6
    pinMode(4, OUTPUT);  // 0 33 -> Epdiy the Data lines I got correctly
    pinMode(5, OUTPUT);  // 1 32
    pinMode(18, OUTPUT);// 2 4
    pinMode(19, OUTPUT);// 3 0
    pinMode(23, OUTPUT);// 4 2
    pinMode(25, OUTPUT);// 5 27
    pinMode(26, OUTPUT);// 6 21
    pinMode(27, OUTPUT);// 7 22

When I see how is the driver has the  GPIOs separated in 3 big groups: 

1. Config register
2. Control lines
3. Data lines (This one I got)

What I'm not copletely sure is where are the CFG GPIOs in Inkplate. 
```
/* Config Register Control */
#define CFG_DATA GPIO_NUM_23
#define CFG_CLK GPIO_NUM_18
#define CFG_STR GPIO_NUM_19

/* Control Lines */
#define CKV GPIO_NUM_25
#define STH GPIO_NUM_26 

#define V4_LATCH_ENABLE GPIO_NUM_15  -> LE Lach enable in Inkplate

/* Edges */
#define CKH GPIO_NUM_5

/* Data Lines */
#define D7 GPIO_NUM_22
#define D6 GPIO_NUM_21
#define D5 GPIO_NUM_27
#define D4 GPIO_NUM_2
#define D3 GPIO_NUM_0
#define D2 GPIO_NUM_4
#define D1 GPIO_NUM_32
#define D0 GPIO_NUM_33
```

Can you please refer me to this Config lines in Inkplate?
As another question I want to ask, this MCP Bus IO expander, has also something to do with the config lines or is just to expand the available GPIOs in the Inkplate?

Video demo with the new TTGO T5S epaper: https://twitter.com/martinfasani/status/1351855551534260225
Thanks in advance to provide me with more information

BornaBiro, 2021-01-25:
There are no config files, but it would be a good idea to put some define macros or comment for connections between paper and ESP32.
In general, you can see how everything is connected in Inkplate 6 [schematic](https://github.com/e-radionicacom/Inkplate-6-hardware/blob/master/Schematics%2C%20Gerber%2C%20BOM/Inkplate6%20Schematics.pdf) but in short:
EPD D0 - GPIO4
EPD D1 - GPIO5
EPD D2 - GPIO18
EPD D3 - GPIO19
EPD D4 - GPIO23
EPD D5 - GPIO25
EPD D6 - GPIO26
EPD D7 - GPIO27
EDP CL - GPIO0
EPD LE - GPIO2
EPD CKV - GPIO32
EPD SPV - GPA2 (MCP23017)
EPD OE - GPA0 (MCP23017)
EPD SPH - GPIO33
EPD GMODE - GPA1 (MCP23017)

TPS65186 PWRUP - GPA4 (MCP23017)
TPS65186 WAKE - GPA3 (MCP23017)
TPS65186 VCOM - GPA5 (MCP23017)

MCP23017 is a classic I/O expander connected with ESP32 through I2C bus. We are using it to save some pins of ESP32 as well as get user more I/O pins for controlling some other hardware. Some of "slow-changing" lines like SPV, GMODE and OE of paper are also connected to I/O expander.

martinberlin, 2021-01-26:
Thanks that was very useful information. I will try to see if I can work out this I2S expander also with esp-IDF and then I can close this.

BornaBiro, 2021-01-26:
Just a note, MCP23017 is a I2C I/O expander not I2S. These are two different communication protocols.

Also, one more thing. EPDiy uses I2S Parallel interface to drive epaper, while we are using classic bit-banging method. In odrer to get fastest refresh rate from our code, we have to optimize it a little bit. That means that we have to set data and clock it in as less instructions as possible. In order to do that, in one line of code, we clear data lines and clock line and in the other we set data and pull clock high. Because you can't set data and clock it at the same time, we have to slow clock down on hardware level and for that we used capacitor from GPIO0 (Epaper CL) to GND (capacitor C44 of 100pF). If you are using EPDiy, that can cause problems. In that case, you should remove it. But, if you ever wanted to use Inkplate library again, you have to put it back.

martinberlin, 2021-01-26:
Ahh that’s very good advice. Thanks a lot!

closed 2021-01-26 by martinberlin