SolderedElectronics/Inkplate-Arduino-library #61: WiFi connection error

davidzovko, 2021-02-16:
@davidzovko  I think examples that don't use wifi looks fine? like the game of life or Inkplate_Low_power looks fine.
the weather example (daily and hourly) ends up with this error : 
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x22 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download
```
which seems like https://github.com/espressif/arduino-esp32/issues/577

weather station seems to work but sometimes reboot
```
E (163717) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (163717) task_wdt:  - IDLE0 (CPU 0)
E (163717) task_wdt: Tasks currently running:
E (163717) task_wdt: CPU 0: wifi
E (163717) task_wdt: CPU 1: loopTask
E (163717) task_wdt: Aborting.
abort() was called at PC 0x400eee43 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008b51c:0x3ffbe360 0x4008b799:0x3ffbe380 0x400eee43:0x3ffbe3a0 0x40082261:0x3ffbe3c0 0x40162086:0x3ffd5df0 0x40155327:0x3ffd5e10 0x4014d22c:0x3ffd5e30 0x4013dc37:0x3ffd5e50 0x4014ff63:0x3ffd5eb0 0x4014aa29:0x3ffd5ed0 0x4014fa49:0x3ffd5f40 0x4014cda4:0x3ffd5f60 0x4014cebe:0x3ffd5f80 0x4015c3d2:0x3ffd5fa0 0x4008c062:0x3ffd5fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10900
load:0x40080400,len:6360
entry 0x400806b4
Serial Monitor Initialized
Welcome to Wol Inkplate 6 weather example!

```

the crypto_tracker example also seems fine but also reboots
the serial monitor has several output like : 
```
E (29030) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (29030) task_wdt:  - IDLE0 (CPU 0)
E (29030) task_wdt: Tasks currently running:
E (29030) task_wdt: CPU 0: wifi
E (29030) task_wdt: CPU 1: IDLE1
E (29030) task_wdt: Aborting.
abort() was called at PC 0x400e88db on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008b540:0x3ffbe4a0 0x4008b7bd:0x3ffbe4c0 0x400e88db:0x3ffbe4e0 0x40082265:0x3ffbe500 0x4015df80:0x3ffdc370 0x4015121b:0x3ffdc390 0x401490cc:0x3ffdc3b0 0x40139aef:0x3ffdc3d0 0x4014be0f:0x3ffdc430 0x401468c5:0x3ffdc450 0x4014b8f1:0x3ffdc4c0 0x40148c44:0x3ffdc4e0 0x40148d5e:0x3ffdc500 0x401582ce:0x3ffdc520 0x4008c086:0x3ffdc550

Rebooting...
ets Jun  8 2016 00:22:57
```

_Originally posted by @indefini in https://github.com/e-radionicacom/Inkplate-Arduino-library/issues/56#issuecomment-779612556_

indefini, 2021-02-16:
after a few hours weather station also stopped with the same error :

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x22 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download
```

nitko12, 2021-02-16:
I'm trying to recreate your bug now, 

could you maybe try compiling your code with esp32 verbose setting, 
maybe this will give us more info on what is going on 🤔 ?

indefini, 2021-02-18:
@nitko12 by "esp32 verbose setting" do you mean in the arduino ide "Tools, Core Debug Level, Verbose" ?

Daily weather station
```
[V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.
W (516298) wifi:Next TBTT incorrect! next beacon:2700379984, beacon interval:102400, now:2704855683
[W][HTTPClient.cpp:811] getStream(): getStream: not connected
[E][WiFiClient.cpp:309] setOption(): fail on fd -1, errno: 9, "Bad file number"
[W][HTTPClient.cpp:811] getStream(): getStream: not connected
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9
[V][HTTPClient.cpp:235] beginInternal(): url: https://www.metaweather.com/api/location/615702/
[W][HTTPClient.cpp:247] beginInternal(): unexpected protocol: https, expected http
[V][HTTPClient.cpp:235] beginInternal(): url: https://www.metaweather.com/api/location/615702/
[D][HTTPClient.cpp:276] beginInternal(): host: www.metaweather.com port: 443 url: /api/location/615702/
[D][HTTPClient.cpp:562] sendRequest(): request type: 'GET' redirCount: 0

[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 181788
[V][ssl_client.cpp:61] start_ssl_client(): Starting socket
[V][ssl_client.cpp:100] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:109] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:165] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:191] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:206] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:227] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:236] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:251] start_ssl_client(): Free internal heap after TLS 141768
[D][HTTPClient.cpp:1108] connect():  connected to www.metaweather.com:443
[V][ssl_client.cpp:289] send_ssl_data(): Writing HTTP request with 169 bytes...
[V][ssl_client.cpp:296] send_ssl_data(): Returning with 169 bytes written
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'x-xss-protection: 1; mode=block'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'Content-Language: en'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'x-content-type-options: nosniff'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'strict-transport-security: max-age=2592000; includeSubDomains'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'Vary: Accept-Language, Cookie'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'Allow: GET, HEAD, OPTIONS'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'x-frame-options: DENY'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'Content-Type: application/json'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'X-Cloud-Trace-Context: cdafcf8534bc8c9951d86e2a7ca8725e'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'Date: Tue, 16 Feb 2021 23:29:14 GMT'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'Server: Google Frontend'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: 'Content-Length: 3469'
[V][HTTPClient.cpp:1199] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1240] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1243] handleHeaderResponse(): size: 3469
[D][HTTPClient.cpp:586] sendRequest(): sendRequest code=200

[D][HTTPClient.cpp:368] disconnect(): tcp keep open for reuse

[V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x22 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download
```

Hourly weather station:
```
D][HTTPClient.cpp:368] disconnect(): tcp keep open for reuse

[V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.
W (22327) wifi:Next TBTT incorrect! next beacon:2709282705, beacon interval:102400, now:2966587275
W (22860) wifi:Next TBTT incorrect! next beacon:2977592899, beacon interval:102400, now:3234920702
W (23393) wifi:Next TBTT incorrect! next beacon:3245972096, beacon interval:102400, now:3503253959
W (23930) wifi:Next TBTT incorrect! next beacon:3514318453, beacon interval:102400, now:3771591432
W (24466) wifi:Next TBTT incorrect! next beacon:3782583155, beacon interval:102400, now:4039927570
W (25001) wifi:Next TBTT incorrect! next beacon:4050920266, beacon interval:102400, now:4308262806
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x22 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download

```

nitko12, 2021-02-18:
Yes, recreated the issue yesterday,
shifting to deepsleep now, could be that lightsleep messes up wifi. 

Will let you know when changes are on master.

nitko12, 2021-02-18:
Here:
https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/dev/examples/Inkplate_6/Projects/Daily_weather_station_example/Daily_weather_station_example.ino

Could you try this version?

It will probably be pushed into master soon.

indefini, 2021-02-24:
@nitko12 this is working but with a few quirks, most of the time only the clock is displayed and the weather is not displaying : looks like partial refresh is not working correctly.
Sometimes it reboots, displaying "welcome to inkplate ...". can I avoid this?
also does this means I should deep sleep every time I want to use the wifi?

nitko12, 2021-02-24:
Yes, it seems deep sleep works better with wifi, there seem to be a bug in retaining connection with light sleep after few hours.

Also, please run the given code with the library in the branch it is in, there was a change in forcing full update after deep sleep wake up.

As for rebooting, it could be that the board can't connect to wifi after given number of times, is there anything useful in the serial monitor after that happens?

indefini, 2021-02-25:
@nitko12 thanks, using the library in the dev branch makes the drawing work fine.

The reboot shows this in the monitor :
```

[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
Waiting for WiFi to connect.....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
..................Can't connect to WIFI, restarting
ets Jun  8 2016 00:22:57
```

davidzovko, 2021-03-01:
Good to hear! It will be implemented into the main branch with the next release.

closed 2021-03-01 by davidzovko

fibis, 2022-03-07:
We are facing this issue again on older Inkplates 6 with the newest library version. Is this maybe more a board problem?