SolderedElectronics/Inkplate-Arduino-library #65: boot loop with 18650

rhousedorf, 2021-02-19:
Hi folks, working well so far, just tried to put it on the battery.

When I insert the battery, or turn on the switch with battery installed, the screen goes black, just like normal, briefly clears to white, then cycles over and over.

Battery is @ 4.18 V
It functions perfectly fine with the battery in and connected to USB.
If I remove the usb cable and re-insert, it reboots as if there was a loss of power.

What am I doing wrong?

davidzovko, 2021-02-21:
Hey,

check [this one](https://github.com/e-radionicacom/Inkplate-Arduino-library/issues/56#issuecomment-751844439), please.

rhousedorf, 2021-02-24:
I put a 4.5 uf  cap in there.  The boot cycle/black screen time is longer, but it still stays in a reboot cycle.

I bought three boards for various projects.  Should I try the other two?

davidzovko, 2021-03-01:
Yes, please try other ones and let me know.

rhousedorf, 2021-03-11:
OK, i've tried all 3 and they all have the same problem.

How do I go about getting them exchanged?

davidzovko, 2021-03-12:
Contact us here to get is solved please: https://e-radionica.com/en/contacts/

closed 2021-03-12 by davidzovko