SolderedElectronics/Inkplate-Arduino-library #69: Inkplate_basic_gray:32:10: error: multi-line comment [-Werror=comment]

as17, 2021-03-05:
the Inkplate_basic_gray has the following line (line 31-33):

`#define DELAY_MS                                                                                                       \`
` 5000 //Delay in milliseconds between screen refresh. Refreshing e-paper screens more often than 5s is not recommended \`
`                //Want to refresh faster? Use partial update! Find example in "Inkplate-basic_partial_update"`

The backslash does not work it throws an error. Changing this to
`#define DELAY_MS 5000 //Delay in milliseconds between screen refresh. Refreshing e-paper screens more often than 5s is not recommended Want to refresh faster? Use partial update! Find example in "Inkplate-basic_partial_update"`
makes this error disappear

nitko12, 2021-03-13:
How did you open and save the file, some autoformatters seem to break multiline precompiler directives?

as17, 2021-03-14:
Hi,
No auto formatting -as I know. Used the Arduino IDE.

But opening the original file in git with the git viewer:

There is this strange new line as well…..
A

closed 2021-03-22 by davidzovko