SolderedElectronics/Inkplate-Arduino-library #70: several errors during compilation

as17, 2021-03-05:
1. I can compile and submit to the Inkplate the example **Inkplate_basic_BW** without any problems.
2. I **cannot** compile many other examples. E.g for **Inkplate_basic_gray** it throws the following error:
`/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp: In member function 'void Inkplate::display3b()':
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:262:82: error: operation on 'dp' may be undefined [-Werror=sequence-point]
             hscan_start((GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]));
                                                                                  ^
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:263:90: error: operation on 'dp' may be undefined [-Werror=sequence-point]
             GPIO.out_w1ts = (GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]) | CL;
                                                                                          ^
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:268:94: error: operation on 'dp' may be undefined [-Werror=sequence-point]
                 GPIO.out_w1ts = (GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]) | CL;
                                                                                              ^
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:270:94: error: operation on 'dp' may be undefined [-Werror=sequence-point]
                 GPIO.out_w1ts = (GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]) | CL;
                                                                                              ^
cc1plus: some warnings being treated as errors
Mehrere Bibliotheken wurden für "WiFiClient.h" gefunden
 Benutzt: /Users/andreas/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Fehler beim Kompilieren für das Board Inkplate 6 (ESP32).`

3. Inkplate_Maze_Generator throws this error:
`/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp: In member function 'void Inkplate::display3b()':
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:262:82: error: operation on 'dp' may be undefined [-Werror=sequence-point]
             hscan_start((GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]));
                                                                                  ^
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:263:90: error: operation on 'dp' may be undefined [-Werror=sequence-point]
             GPIO.out_w1ts = (GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]) | CL;
                                                                                          ^
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:268:94: error: operation on 'dp' may be undefined [-Werror=sequence-point]
                 GPIO.out_w1ts = (GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]) | CL;
                                                                                              ^
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/boards/Inkplate6.cpp:270:94: error: operation on 'dp' may be undefined [-Werror=sequence-point]
                 GPIO.out_w1ts = (GLUT2[k * 256 + (*(--dp))] | GLUT[k * 256 + (*(--dp))]) | CL;
                                                                                              ^
cc1plus: some warnings being treated as errors
Mehrere Bibliotheken wurden für "WiFiClient.h" gefunden
 Benutzt: /Users/andreas/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Fehler beim Kompilieren für das Board Inkplate 6 (ESP32).?` as well
4. the Game_Of_Life_by_claud9999 returns an other error:
`/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/examples/Inkplate_6/Community_Contributions/Game_Of_Life_by_claud9999/Game_Of_Life_by_claud9999/Game_Of_Life_by_claud9999.ino: In function 'void loop()':
Game_Of_Life_by_claud9999:126:32: error: suggest parentheses around '&&' within '||' [-Werror=parentheses]
             if (cell_next != 0 && cell_curr == 0 || cell_next == 0 && cell_curr != 0)
                                ^
cc1plus: some warnings being treated as errors
Mehrere Bibliotheken wurden für "WiFiClient.h" gefunden
 Benutzt: /Users/andreas/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
suggest parentheses around '&&' within '||' [-Werror=parentheses]`

As if Arduino does not understand the code correctly.......
Any Idea?

Setup:
Arduino IDE v 1.8.13; python 2.7.16; python3 3.9.2; MacMini M1 (2020); BigSur 11.2.2; Inkplate6; installed Inkplate board definition and example files today.

nitko12, 2021-03-13:
Could you try with the new code on master?

It was indeed a warning that your IDE probably considered as an error.

as17, 2021-03-14:
Hi

I tried the following examples (Inkplate6) using the git library https://github.com/e-radionicacom/Inkplate-Arduino-library.git <https://github.com/e-radionicacom/Inkplate-Arduino-library.git> downloaded today:
(I am using the native Arduino IDE - no other tool that does line breaks!)

Hope this helps a little……

MySetup: BigSur 11.2.3, Xcode command line tool installed, Arduino 1.8.13

You did what I always wanted to do: a nice clean eInkPaper monitor!!!!! Cool stuff!

Andreas.

Inkplate_basic_gray:
ERROR
error with line#31 (multiline comment); after correcting - NO problem anymore
WARNING
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/include/ShapesPolygon.cpp: In member function 'void Shapes::fillPolygon(int*, int*, int, int)':
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/include/ShapesPolygon.cpp:239:45: warning: 'y2' may be used uninitialized in this function [-Wmaybe-uninitialized]
             storeEdgeInTable(x1, y1, x2, y2);
                                             ^
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/src/include/ShapesPolygon.cpp:239:45: warning: 'x2' may be used uninitialized in this function [-Wmaybe-uninitialized]

=> WORKS (after correcting error)

Inkplate_basic_BW:
ERROR
error with line#26 (multiline comment); after correcting - NO problem anymor
WARNING
—> WARNING(Inkplate_basic_gray)

=> WORKS (after correcting error)

Inkplate_basic_partial_update:
ERROR: none
WARNING
—> WARNING(Inkplate_basic_gray)

=> WORKS

Inkplate_basic_partial_update:
ERROR: none
WARNING
—> WARNING(Inkplate_basic_gray)

=> WORKS

Inkplate_Battery_Voltage_And_Temperature:
ERROR: none
WARNING
—> WARNING(Inkplate_basic_gray)

=> WORKS

Game_Of_Life_by_claud9999:
ERROR: 
/Users/andreas/Documents/Arduino/libraries/InkplateLibrary/examples/Inkplate_6/Community_Contributions/Game_Of_Life_by_claud9999/Game_Of_Life_by_claud9999/Game_Of_Life_by_claud9999.ino: In function 'void loop()':
Game_Of_Life_by_claud9999:126:32: error: suggest parentheses around '&&' within '||' [-Werror=parentheses]
             if (cell_next != 0 && cell_curr == 0 || cell_next == 0 && cell_curr != 0)
                                ^
cc1plus: some warnings being treated as errors
Mehrere Bibliotheken wurden für "WiFiClient.h" gefunden
 Benutzt: /Users/andreas/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi

exit status 1
suggest parentheses around '&&' within '||' [-Werror=parentheses]
WARNING

=> DOES NOT WORK
[... 126 lines cut ...]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:104:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:104:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:104:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:104:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:104:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:104:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:104:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:120:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 };
 ^
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:120:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/var/folders/k_/76ykyjh54ps54wd787xbtcww0000gn/T/arduino_modified_sketch_346362/Crypto_tracker_example.ino:120:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
Mehrere Bibliotheken wurden für "WiFiClient.h" gefunden
 Benutzt: /Users/andreas/Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi

=> does NOT WORK

Inkplate_VariPass_Graphs:
	
ERROR: none
WARNING
—> WARNING(Inkplate_basic_gray)

=> WORKS

nitko12, 2021-03-15:
Hi, 

multiline comments have been removed, I'll try removing others soon.

You could try changing compile warnings to none in Arduino prefrences for now, but the fix is coming for -Wall too.

I've opened an issue on the site where fonts we're generated, but will replace them with warning free ones soon too.

as17, 2021-03-16:
Thanks!
I can work with the library as is now. The functionalities I need work….
CU

nitko12, 2021-07-06:
In the latest push to dev branch I've removed all warnings I've managed to find, in the next release it should all work even with all warning set in the IDE :) 

I'll close this now, feel free to reopen if you find anything weird again :D

closed 2021-07-06 by nitko12