SolderedElectronics/Inkplate-Arduino-library #73: Incorrect ZIP archive

debackerl, 2021-04-08:
Hello,

I'm using arduino-cli on Linux. To install this library I had been trying

```
wget https://github.com/e-radionicacom/Inkplate-Arduino-library/archive/refs/tags/2.2.0.zip
arduino-cli lib install --zip-path 2.2.0.zip
```

But, arduino-cli returned an error (2nd line):
```
--git-url and --zip-path flags allow installing untrusted files, use it at your own risk.
Error installing Zip Library: library is not valid: missing header file "Inkplate-Arduino-library-2.2.0.h"
```

This comes from arduino indeed: https://github.com/arduino/arduino-cli/blob/5f4f6e90a65e29c0670da9794f6e35714931fc23/arduino/libraries/librariesmanager/install.go#L264

Fixing the problem is easy, inside the Zip file, instead of calling the directory Inkplate-Arduino-library-2.2.0, it must be Inkplate-Arduino, and fair enough it worked:

```
unzip 2.2.0.zip
mv Inkplate-Arduino-library-2.2.0 Inksplate
zip -r Inksplate.zip Inksplate
arduino-cli lib install --zip-path Inkplate.zip
```

And it worked:

```
--git-url and --zip-path flags allow installing untrusted files, use it at your own risk.
Installed Archived Library
```

I suppose that I may not be the only one trying to doing your official ZIP releases, and simply trying to install it with arduino-cli. If you could use the proper directory name, then it would work out of the box.

Disclaimer: I'm not using Arduino a lot, so maybe there is something deeper going there.

Thank you,
Laurent

PericR, 2021-04-10:
Hello, thank you for contacting us here. I see how this can cause problems. We should have new release soon and I'm gonna see that we implement new practice for naming tags.

PericR, 2021-04-12:
Hi, so that directory is automatically named by github when you download using that link. For downloading, try to use this link:
https://github.com/e-radionicacom/Inkplate-Arduino-library/archive/refs/heads/master.zip and let me know what happens.

PericR, 2021-04-22:
I'm closing this due to inactivity. Please contact us if you need more information's.

closed 2021-04-22 by PericR