SolderedElectronics/Inkplate-Arduino-library #76: Daily weather station example uses partialUpdate after deep sleep

spaetz, 2021-04-21:
Hi, great examples in the library. I was experimenting with the Daily Weather Station one and wondered why the whether was deleted and only the clock shown most of the time.

I discovered that in most updates after waking from deep sleep you only update the time and call  partialUpdate()

https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/05216b92c330918ee284a0fac50a32047439fe95/examples/Inkplate6/Projects/Daily_weather_station_example/Daily_weather_station_example.ino#L177

However, after deep sleep that clears the screen and all the weather information is gone. I doubt this is the intended behavior, so I am filing this issue to make people aware.

PericR, 2021-04-22:
Hello, I'll take a look at that soon. you can make it work by removing else part in setup until we fix it.

spaetz, 2021-04-22:
Yes, I fixed it by enforcing a full refresh on each cycle. But this will of course fetch the full whether data on each cycle which I want to avoid.

But it is very nice and easy to understand code thanks. I extended it by manually configuring a woeid (location id from the whether data API), so I don't have to look up my city name in each run...

PericR, 2021-04-23:
Trick is to save data variables with RTC_DATA_ATTR and then redraw it all on partial update.

PericR, 2021-04-26:
Hello, I fixed problem with partial update part in this example. Please refer to this link since its not on master yet.
[https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/dev/examples/Inkplate6/Projects/Daily_weather_station_example/Daily_weather_station_example.ino](url)

spaetz, 2021-04-26:
Wow, that was a quick fix. And an elegant solution, thanks. I love my inkplate! :)

closed 2021-04-26 by PericR