SolderedElectronics/Inkplate-Arduino-library #80: A little bug in (almost) all src/boards devices initialising the 2nd MCP

turgu1, 2021-05-15:
Hello,

The following lines:

```
    for (int i = 0; i < 15; i++)
    {
        pinModeInternal(MCP23017_EXT_ADDR, mcpRegsInt, i, OUTPUT);
        digitalWriteInternal(MCP23017_EXT_ADDR, mcpRegsInt, i, LOW);
    }
```

should need to be replaced with:

```
    for (int i = 0; i < 15; i++)
    {
        pinModeInternal(MCP23017_EXT_ADDR, mcpRegsEx, i, OUTPUT);
        digitalWriteInternal(MCP23017_EXT_ADDR, mcpRegsEx, i, LOW);
    }
```

Cheers!
Guy

PericR, 2021-05-17:
Hey Guy, thanks for finding this.
The way MCP functions work, it doesn't matter which one is used for initialization and it works fine either way, however it is a bug and I have changed it for the sake of code readability :)

https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/1636f72a1a04d48c0910fa0ba78f8e925967d5ac

closed 2021-05-17 by PericR

turgu1, 2021-05-17:
> Hey Guy, thanks for finding this.
> The way MCP functions work, it doesn't matter which one is used for initialization and it works fine either way, however it is a bug and I have changed it for the sake of code readability :)
> 
> [1636f72](https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/1636f72a1a04d48c0910fa0ba78f8e925967d5ac)

Hello Peric,

That's why I called it a "Little bug". It would be more of an issue for a third party that would requires the use of the 2nd MCP and assume that the Register variables where initialized.

Cheers!
Guy