SolderedElectronics/Inkplate-Arduino-library #81: What type of pngs are supported?

dolanmiu, 2021-05-16:
I don't understand what criteria are needed for a successful PNG to be displayed from the web

I can't see a pattern here. I have ruled out the bit depth because bit depth of 32 works

Can someone help?

## Works:

```cpp
display.drawImage("https://compresspng.com/images/compresspng/icon.png", Image::PNG, 0, 0, true, false)
display.drawImage("https://assets.stickpng.com/images/580b57fcd9996e24bc43c51f.png", Image::PNG, 0, 0, true, false)
display.drawImage("https://cdn.pixabay.com/photo/2015/12/11/11/43/google-1088003_960_720.png", Image::PNG, 0, 0, true, false)
display.drawImage("http://openweathermap.org/img/wn/02n@4x.png", Image::PNG, 0, 0, true, false)
```

## Doesn't work:

```cpp
display.drawImage("https://www.learningcontainer.com/wp-content/uploads/2020/08/Sample-Webp-File-for-Testing.png", Image::PNG, 0, 0, true, false)
display.drawImage("https://lh3.googleusercontent.com/proxy/tvO9KuzGPjesoQ_Ym37lFzy64bkYaD7N3AaclvQY2UtXRScaVRj0EWNxYtqWyicGkgbREMSa8K666PnTqVlUm3W-npLbegHDu6bVasK117M5", Image::PNG, 0, 0, true, false)
display.drawImage("https://i.pinimg.com/originals/fc/de/94/fcde946d572f8968cde14b688527021b.png", Image::PNG, 0, 0, true, false)
```

PericR, 2021-05-21:
Hello, I made a patch in our function for file downloading and managed to fix third image. For the second one I'm getting 403 forbidden. 
Also can you confirm that first one is in right format. You can only use Windows Bitmap file with color depth of 1, 4, 8 or 24 bits with no compression!

dolanmiu, 2021-05-21:
First image:
https://www.learningcontainer.com/wp-content/uploads/2020/08/Sample-Webp-File-for-Testing.png

Yes correct second image is a 403 it seems. It was working at the time of writing

dolanmiu, 2021-05-21:
What tool do you use to confirm that a PNG is of the correct colour depth?

PericR, 2021-05-25:
> What tool do you use to confirm that a PNG is of the correct colour depth?

Sorry, I thought of bmp, my bad.

PericR, 2021-05-25:
> First image:
> https://www.learningcontainer.com/wp-content/uploads/2020/08/Sample-Webp-File-for-Testing.png

After a bit of digging trough library I find out that this one returns incorrect png signature, causing the  error.  If this is your site, can you try to swap it with another one?

PericR, 2021-06-07:
Closing this due to inactivity, please message me if you need anything else.

closed 2021-06-07 by PericR