SolderedElectronics/Inkplate-Arduino-library #85: Bluetooth Example?

mcwhittemore, 2021-07-10:
Thanks for making the Inkplate, it's been pretty great to use so far. I've been working through the demos getting aquatinted and noticed that there aren't any examples with the bluetooth features of the ESP32. Has there been discussion about adding one? I'm running into errors trying to merge the espressif bluetooth examples and a basic "hello world" and it leaves me wondering if there is something special I need to do in this context that I don't need to do on other ESP32 boards I have.

nitko12, 2021-07-12:
Hi,

heres a quick example we'll merge into master with the next update.

So our board definition can't use bluetooth right now, but we'll fix that soon.

On Inkplate 6 you can use generic ESP32 Wroover board definition and BT will work fine, as in the example here:

https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/d5f9feeb3634419cd036747988687c8b0cb3064c

Will let you know when this is fixed in the board defs :)

mcwhittemore, 2021-07-13:
Thanks @nitko12. I was able to get a pretty simple demo up in running by moving to `esp32:esp32:esp32wrover:UploadSpeed=115200` for my FQBN.

nitko12, 2021-07-13:
Awesome :)

Will close this now, feel free to let us know if you need anything else.

closed 2021-07-13 by nitko12