SolderedElectronics/Inkplate-Arduino-library #92: fast refresh mode

zhangtemplar, 2021-07-24:
Look at the implementation of display or display3b in src/boards/Inkplate10.cpp, I realized the display functions involves some bit operation on data saved to DMemory4Bit and write GPIO.out_w1ts and GPIO.out_w1tc. I assume those are I2C pins.

So my question is that could we achieve fast refresh of display via:
- provide data to display already aligned to the bit operation;
- use DMA/PWM to transfer the data via GPIO.out_w1ts and GPIO.out_w1tc.

Is this possible? What will be the highest refresh rate?

The potential applications is second monitor or a monitor to drop a sreenshot.

BornaBiro, 2021-07-26:
Inkplate panel doesn't use any popular communication protocol such as I2C. It uses a dedicated protocol for bare epaper panels. To make this possible to drive with ESP32, communication has been made by bit-banging. DMemory4Bit is just a frame buffer for the current image on the screen that stores two 4 bit pixels and GPIO.out_w1ts and GPIO.out_w1tc are just registers for GPIO direct port manipulation for faster bit-banging.
Providing aligned data in frame buffer to the GPIO is not possible because to get some shade of gray, a special combination of black and white pixels has to be sent for each frame (and one refresh consists of multiple frames). This is called [waveform](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/boards/Inkplate10.h#L85) and we actually optimized code for faster refresh by using LUTs [here](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/boards/Inkplate10.cpp#L273) and [here](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/boards/Inkplate10.cpp#L122).
I really don't see how PWM and DMA can speed up refresh, especially because there is no DMA for GPIO in ESP32.
There are few things that can speed up refresh rate to some extend and we are currently doing some experiments with that. You have to bare in mind that all these panels have physical limitations when comes to refresh rate.

zhangtemplar, 2021-07-27:
I see. Thanks for the information.

closed 2021-07-27 by zhangtemplar