SolderedElectronics/Inkplate-Arduino-library #96: Example with international fonts

honzab, 2021-08-22:
Hello,

I am trying to display non-ascii text on my Inkplate10. Is there an example somewhere around how to handle or include international fonts? All that I could find include fonts with only ascii characters defined.

Thanks!
/Jan

nitko12, 2021-08-23:
Hi,

what characters do you need exactly?

I searched is something like this possible and found this GH issue.
https://github.com/adafruit/Adafruit-GFX-Library/pull/200

So Adafruit GFX does not support it currently, bu there is Bodmers proof of concept fix,
I don't think we'll include this but you could do it yourself, just replace our Adafruit GFX files in the library with these

https://github.com/Bodmer/Adafruit-GFX-Library

And it should work fine, if you need more help about how to convert using fontconvert tool feel free to ask here :)

honzab, 2021-08-23:
Mostly characters from different european languages - öäåçčćéěë etc.

I am receiving data as JSON a can print the correct representation to the serial console. Being fairly new to embedded development, I looked first at the font header files and noticed that they do not define the characters.

However, with what you are writing and linking, it looks like it wouldn't "just work" if I managed to get a font file that defines those characters?

nitko12, 2021-08-24:
Hmm some of those could work,

ascii value of these chars seems to be 246, 228, 229, 231, 269, 263, 233, 283, 235

and if it's under 255 it could work out of the box, haven't tested tho.

There could be another workaround to map these to some other characters in basic ascii, but that ain't a permanent solution.
Supporting other charsets is something I doubt Adafruit will ever add to GFX :)