SolderedElectronics/Inkplate-Arduino-library #97: Provide example and/or way to disable battery led on inkplate 10

jakobwesthoff, 2021-08-23:
I am currently working on a battery powered project utilizing an Inkplate 10. In order to minimize power consumption, I am of course using the sleep functionality of the esp32. However I noticed that there is a battery indicator led near the on/off switch on the board, which is *only* active is a battery is attached.

In order to minimize power consumption, which is why, I chose this epaper style display in the first place ;), I don't want to waste quite some power on a quite bright led.

Is there anyway to disable this led? If so, could you provide an example an example or point me into the right direction?

I wasn't able to find something regarding this. However I won't believe this was an oversight on your part and the only way to disable this led would be to remove it from the board entirely.

Thanks in advance and congratulations on an otherwise beautiful and amazing product.

davidzovko, 2021-08-24:
Hi @jakobwesthoff! Thanks for your buying Inkplate and your kind words :) 

I 100% agree, any unneeded power draw must be removed from the board. We thought of that, and by design, the board should have its battery charging indicator LED turned off when there is no USB power, but there is a battery. I just tested that:
With battery, no USB power:
[image]

With battery AND USB power:
[image]

Therefore, the LED status is expected and will not draw additional power when battery-powered + in sleep. Can you please check your board, does it work differently?

jakobwesthoff, 2021-08-25:
That is exactly what I thought should happen, as I assumed the `CHRG` indicator should only indicate charging ;).

However I have the following effects here, with and without a usb cable attached:

[image]
[image]

But after hearing from you, that this behavior is not normal I started testing different things. One of them exchanging the attached battery for another one. And suddently the led is off. I have no idea how actually the battery itself could cause this. As I can reproduce the behavior with exactly one of my lion cells (I tested with three different ones). 

[image]

Nevertheless I will discard the battery which seems to malfunctioning and simply write this off as an electronics fluke ;).

Thanks for getting back to me this quick and for the help. Maybe having documented this phenomenon does help others with equal problems in the future.