SolderedElectronics/Inkplate-Arduino-library #98: USB or BT Keyboard Example

deanhuff, 2021-09-07:
do you have any examples that would show key input to the Inkplate 6 via either a bluetooth or USB connected keyboard?

davidzovko, 2021-09-08:
Hi, for wired connection, I am afraid that you need external hardware. USB Host board or breakout or similar would be the easiest, although I have found this as well: https://github.com/sdima1357/esp32_usb_soft_host
If you have PS2 keyboard laying around, you might try this as well: https://github.com/michalhol/ps2kbdlib

That's cool application for Inkplate. We have been preparing an option where partialUpdate() does not turn off the power supply between each partial refresh, thus making it much faster. I will check on that for you..

deanhuff, 2021-09-08:
I would very much like to try out the option where the displayPartial() function operates faster.

I will look into adding a usb host board.

The comments at the top of the Bluetooth example for the Inkplate 6 state that you must use a different board definition.  Is this still the case?

davidzovko, 2021-09-08:
With this release, you should be able to access faster refresh times: https://github.com/e-radionicacom/Inkplate-Arduino-library/releases/tag/5.2.0
You can check how it's used here: https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/aeaf617e05cdfcd790d6900e56d20ab2e51a4b63/examples/Inkplate6/Basic_Inkplate_Functionality/Inkplate_Basic_Partial_Update/Inkplate_Basic_Partial_Update.ino#L67

As for the USB, I wouldn't say you need another board definition. You can use this one with proper BT example.

deanhuff, 2021-09-08:
I'm not really seeing any difference in partialUpdate speeds when comparing 5.2.0 vs 5.0.0 libraries.  I am not sure that I installed the 5.2.0 library correctly.  here's what I did:
  1. I went to ~/Arduino/libraries and deleted anything that said Inkplate.  
  2. next I did 'tar -xzf ~/Downloads/Inkplate-Arduino-library-5.2.0.tar.gz' in the libraries folder.  
  3. I recompiled my program so that it writes a row of letter 'a' across the screen.  

Both seemed to take about the same amount of time using either library.

Here is my code:

```
#include <Inkplate.h>
Inkplate display(INKPLATE_1BIT); 

// Include fonts used
#include "Fonts/FreeSansBold12pt7b.h"
int tsize=1;

void setup() {
    //Serial.begin(115200);
    //Serial.println("testing");
  
    display.begin();        // Init library (you should call this function ONLY ONCE)
    display.clearDisplay(); // Clear any data that may have been in (software) frame buffer.
    //(NOTE! This does not clean image on screen, it only clears it in the frame buffer inside ESP32).
    display.display(); // Clear everything that has previously been on a screen
    display.setCursor(2,25);
    display.setRotation(1);
    
    display.setTextWrap(true);
    display.setTextSize(tsize);
    display.setTextColor(BLACK, WHITE);
    display.setFont(&FreeSansBold12pt7b);
    display.print("TEST: Picowriter on Inkplate 6!\n");
    display.display(); // Write hello message
    
    //Serial.setTimeout(1); // set serial timeout to something really low.
    display.einkOn();

    for(int i=0; i<42; i++){ // testing write 1 row of 'a' as fast as possible
      display.print('a');
      display.partialUpdate();
    }
    display.einkOff();
}

void loop()
{
}
```

davidzovko, 2021-09-15:
Sorry, I gave you the wrong function to use. You need to use this one: https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/boards/Inkplate6.cpp#L360

By using `display.paritalUpdate(0,1);` you will achieve the shortest possible refresh time. 

Let me know how that worked out for you!

deanhuff, 2021-09-15:
indeed, this is dramatically faster.  for my test of writing a row of writing a row of 'a' i experienced the following times (from system on by toggling the on/off switch).

- display.partialUpdate() 19.98 seconds
- display.partialUpdate(0,1) 12.65 seconds

I'll get the USB HOST stuff figured out. 

Thanks for your help, closing this out.

closed 2021-09-15 by deanhuff

davidzovko, 2021-09-15:
Good to hear, good luck with the typewriter!

dedesite, 2021-10-30:
Hi @deanhuff I also want to create a typewriter with an inkplate 10 and be very pleased to hear about the solution you found for USB keyboard.
Also, would be cool to have a basic text editor code.
Think about a Markdown editor with a way to navigate easily between headers.

dedesite, 2021-10-30:
Oh sorry if I pollute" this issue, but I've seen on reddit that partial update on the inkplate 10 are slower than the inkplate 6, is it still the case with the new firmware ?

deanhuff, 2021-10-30:
My project hasnt made much progress.  Hopefully I'll get motivated soon to get back on it

dedesite, 2021-11-07:
Hello, I've trying to pair a BT 3.0 (no BLE so) keyboard with my inkplate 6 without success.
I used the "ESP32 Wroover Module" board from this definition : https://dl.espressif.com/dl/package_esp32_index.json

And while `Serial.begin` works with this board definition (otherwise with the inkplate it was always rebooting) I can't connect to my keyboard.

Here is the code : 
```cpp

#include "Inkplate.h"
#include "Arduino.h"
#include "Fonts/Open_Sans_SemiBold_32.h"
#include "BluetoothSerial.h"

const GFXfont *text2_font = &Open_Sans_SemiBold_32;

BluetoothSerial SerialBT; //Object for Bluetooth

String MACadd = "DC:2C:26:E4:67:D6";
uint8_t address[6]  = {0xDC, 0x2C, 0x26, 0xE4, 0x67, 0xD6};
String name = "Bluetooth USB keyboard";

Inkplate display(INKPLATE_1BIT);

int incoming;
bool connected;

void setup()
{
    display.begin();        // Init library (you should call this function ONLY ONCE)
    display.clearDisplay(); // Clear any data that may have been in (software) frame buffer.
    //(NOTE! This does not clean image on screen, it only clears it in the frame buffer inside ESP32).
    display.display(); // Clear everything that has previously been on a screen
    display.setCursor(150, 320);
    display.setTextSize(1);
    display.setFont(text2_font);

    // Tentative BT
    Serial.begin(9600); //Start Serial monitor in 9600
    SerialBT.begin("ESP32_KEYBOARD", true); //Name of your Bluetooth Signal
    display.print("Bluetooth Device is Ready to Pair");
    display.display();
    // connect(address) is fast (upto 10 secs max), connect(name) is slow (upto 30 secs max) as it needs
    // to resolve name to address first, but it allows to connect to different devices with the same name.
    // Set CoreDebugLevel to Info to view devices bluetooth address and device names
    //connected = SerialBT.connect(address);
    connected = SerialBT.connect(name);

    if(connected) {
      display.println("Connected Succesfully!");
      display.partialUpdate(false, true);
    } else {
      while(!SerialBT.connected(10000)) {
        display.print("Failed to connect.");
        display.partialUpdate(false, true);
      }
    }
}

void loop()
{
 
}
```
I tested with `address` and `name` without success. I checked the address, it is good.

Also try to set CoreDebugLevel to Info, but didn't find where those debug info are displayed ?

Do you have any idea on why this fails ?

Thanks,
Andréas

BornaBiro, 2021-11-08:
Hi Andréas,

we checked Your code and indeed, it falis to connect to slave device.
We even tried multiple versions of official ESP32 cores and it also won't connect to it. Master mode works without a problem.

We will further investigate this issue.

Thank You for letting us now about this problem.

dedesite, 2021-11-08:
Hi @BornaBiro ,
Thanks for your reply.
I'm quite new to BT and microcontroller dev, can you confirm me that I can't connect a keyboard in master mode ?
It's obviously in slave mode since it can't pair devices right ?
Hope you'll find a solution, meanwhile I'll try to use the keyboard with software USB implementation, but don't think it will work, because my keyboard is USB 2.0 Full speed...

dedesite, 2021-11-17:
Hi,

I still try to get a USB Full Speed keyboard working with the inkplate 6.
@davidzovko you mention that we need an external board to get USB on the Inkplate.
Would the arduino USB host borad works ? https://docs.arduino.cc/retired/shields/arduino-usb-host-shield

Or is there something else you would recommend ?

Thanks,
Andréas

davidzovko, 2021-11-17:
Hey, yes, that would work! As the matter of fact, any USB host should work, but take notice that they are SPI, so you need to connect to Inkplate SPI pins and use MCP23017 IO pins as CS (Chip Select) pin, since there are no available pins on the ESP32.

deanhuff, 2021-11-17:
Andreas,

Please post a followup of how your USB works out, I'm interested in doing the same.

dedesite, 2021-11-17:
@davidzovko thanks for your reply.
@deanhuff I don't know if I have the competence to make such wiring, I'm really new to electronic...
I'm hesitating on using a PS2 (because it seems simpler to use), but it's difficult (impossible ?) to find a 60% PS/2 keyboard... I want to do a setup similar as the ultimate writer. I'll have to make a choice :).

dedesite, 2021-11-24:
Hi @davidzovko, I bought a Mini USB Host shield which [seems to work fine on ESP32](https://github.com/espressif/arduino-esp32/issues/1271).
I try to follow [a french tutorial](https://wikifab.org/wiki/Brancher_des_p%C3%A9riph%C3%A9riques_USB_sur_un_ESP8266) but got stuck on finding where are SPI MOSI and MISO pins. After looking at the [hardware schematics](https://github.com/e-radionicacom/Inkplate-6-hardware/blob/master/Schematics%2C%20Gerber%2C%20BOM/v1.0/Inkplate6%20Schematics.pdf) MISO and MOSI are in K11 and K12, but I don't understand what does that mean ?
I see that there are "weveform SPI flash" pins, but they are not labeled MOSI or MISO.

I also didn't found the "RESET" pin which seems to be needed for the shield...

Sorry if I don't understand all your schematics, I'm not used to do this sort of things.

And what do you mean by chip select pins ?

@deanhuff if you have more experience than me in this field, maybe we can help each others, my project is hosted here : https://framagit.org/Andreas/e-typewriter/

Will keep looking for answers, thanks,
Andréas

dedesite, 2021-11-24:
Sorry for flooding, but here what I understand and want to try in order to get a Mini USB Host Shield (which is a SPI device) working with Inkplate 6 : 

MOSI / SPI - SDI
MISO / SPI - SDO
CLOCK/ SPI - SCL
CS / SPI - NCS
RST / MCP23017 - INTA

Am I right ?
Since I got confirmation, I'll try playing with it and hopefully get a usb keyboard working :).

BornaBiro, 2021-11-25:
K11 and K12 are header pins on edge of the Inkplate board with the "ESP32" label above them.
SPI pins are:
MOSI - GPIO13
MISO - GPIO12
SCK - GPIO14
Bear in mind that these pins are shared with SD card.

Waveform SPI flash are pins from epaper FLASH memory IC that store waveforms, they are not connected to the ESP32 SPI pins, so do not use them for USB Host.

SPI and I2S are two different communication protocols, you can learn more about them [here](https://learn.sparkfun.com/tutorials/serial-peripheral-interface-spi/all) and [here](https://learn.sparkfun.com/tutorials/i2c/all). I assume that you are using MAX3421E and by looking in datasheet, it uses SPI communication protocol.

For CS pin, you can use GPB5, GPB6 or GPB7 from MCP23017 (GPG2 is not a good choice, because it's shared with touchpads), but keep in mind that you will have to modify MAX3421E library in order to work with MCP23017 (it does not use classic digitalWrite() functions).

dedesite, 2021-11-25:
@BornaBiro Thanks a lot for your anwsers.
I got it all wrong :face_with_spiral_eyes: .

I learn all this stuff in the same time, so it's a bit difficult !

> K11 and K12 are header pins on edge of the Inkplate board with the "ESP32" label above them.
SPI pins are:
MOSI - GPIO13
MISO - GPIO12
SCK - GPIO14

Is there a place where all those informations are written ? 
Maybe it will be a good addition to the schematics in the official documentation ?

> Bear in mind that these pins are shared with SD card.

Hum, what will be the consequences ? For example, could it interfere with my keyboard or corrupt data if I write on SD when typing on the keyboard or something like that ? As I understood, SPI protocole can manage several slaves at the same time no ?

> SPI and I2S are two different communication protocols, you can learn more about them here and here. I assume that you are using MAX3421E and by looking in datasheet, it uses SPI communication protocol.

Yes I was aware of that, but didn't know if I could use some I2C pins for other purpose, but it seemed a silly idea.

> For CS pin, you can use GPB5, GPB6 or GPB7 from MCP23017 (GPG2 is not a good choice, because it's shared with touchpads),

:+1: 

> but keep in mind that you will have to modify MAX3421E library in order to work with MCP23017 (it does not use classic digitalWrite() functions).

Hum, by "MAX3421E library" you mean [this library](https://github.com/felis/USB_Host_Shield_2.0) or something else ?

Thanks again, it help me A LOT !

dedesite, 2021-11-27:
Hi,
I try to adapt USB Host Shield 2.0 library to Inkplate and got another question, which PIN should I use for the USB host shield INT pin ?
MCP23017 - INTA ? Or something else ?
On a ESP32 dev board, it is GPIO17 (TX2).

Thanks,
Andréas