SolderedElectronics/Inkplate-Arduino-library #100: Using SPIFFS and Inkplate.h does not work

michaelotto, 2021-09-10:
If I use Inkplate.h and FS/SPIFFS in the same sketch, like this

#include "Inkplate.h"               // Include Inkplate library to the sketch
#include "FS.h"
#include "SPIFFS.h"

I get a namespace clash error:

```
../Arduino/libraries/InkplateLibrary/src/include/../libs/SdFat/FatLib/ArduinoFiles.h:122:7: error: redefinition of 'class fs::File'
 class File : public FatFile, public Stream {
```

nitko12, 2021-09-22:
Can you please provide the code in question?

The bug is probably beacuse there is a class with the same name in the Inkplate library, I'll add a define or something where you can make the compiler ignore it.

michaelotto, 2021-09-22:
```
#include "Inkplate.h"
#include "FS.h"
#include "SPIFFS.h"

void setup() { }
void loop() { }
```
=>

```
In file included from .../test/test.ino:2:0:
.../Library/Arduino15/packages/Croduino_Boards/hardware/Inkplate/1.0.1/libraries/FS/src/FS.h:118:11: error: 'File' is already declared in this scope
 using fs::File;
```

ChallengesNotPromises, 2021-09-24:
Michael

Thank you for the quick reply.
Here is the sketch I used. I received an email to an updated repository so I will see where that goes soon and try to get it working.

Joseph

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

ChallengesNotPromises, 2021-09-24:
Before I decide to update the sketch, I let it run until it displayed correctly. Here is the serial monitor output. The successful one is the last one at the bottom.

Joseph

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

00:59:16.764 -> ets Jun  8 2016 00:22:57
00:59:16.764 -> 
00:59:16.764 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
00:59:16.764 -> configsip: 0, SPIWP:0xee
00:59:16.810 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:59:16.810 -> mode:DIO, clock div:2
00:59:16.810 -> load:0x3fff0018,len:4
00:59:16.810 -> load:0x3fff001c,len:1100
00:59:16.810 -> load:0x40078000,len:10900
00:59:16.810 -> load:0x40080400,len:6360
00:59:16.810 -> entry 0x400806b4
01:04:18.477 -> ets Jun  8 2016 00:22:57
01:04:18.477 -> 
01:04:18.477 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
01:04:18.525 -> configsip: 0, SPIWP:0xee
01:04:18.525 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:04:18.525 -> mode:DIO, clock div:2
01:04:18.525 -> load:0x3fff0018,len:4
01:04:18.525 -> load:0x3fff001c,len:1100
01:04:18.525 -> load:0x40078000,len:10900
01:04:18.525 -> load:0x40080400,len:6360
01:04:18.525 -> entry 0x400806b4
01:09:20.126 -> ets Jun  8 2016 00:22:57
01:09:20.126 -> 
01:09:20.126 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
01:09:20.126 -> configsip: 0, SPIWP:0xee
01:09:20.126 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:09:20.126 -> mode:DIO, clock div:2
01:09:20.126 -> load:0x3fff0018,len:4
01:09:20.173 -> load:0x3fff001c,len:1100
01:09:20.173 -> load:0x40078000,len:10900
01:09:20.173 -> load:0x40080400,len:6360
01:09:20.173 -> entry 0x400806b4
01:14:20.731 -> ets Jun  8 2016 00:22:57
01:14:20.731 -> 
01:14:20.731 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
01:14:20.731 -> configsip: 0, SPIWP:0xee
01:14:20.777 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:14:20.777 -> mode:DIO, clock div:2
01:14:20.777 -> load:0x3fff0018,len:4
01:14:20.777 -> load:0x3fff001c,len:1100
01:14:20.777 -> load:0x40078000,len:10900
01:14:20.777 -> load:0x40080400,len:6360
01:14:20.777 -> entry 0x400806b4
01:19:21.888 -> ets Jun  8 2016 00:22:57
01:19:21.888 -> 
01:19:21.888 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
01:19:21.888 -> configsip: 0, SPIWP:0xee
01:19:21.888 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:19:21.888 -> mode:DIO, clock div:2
01:19:21.888 -> load:0x3fff0018,len:4
01:19:21.888 -> load:0x3fff001c,len:1100
01:19:21.888 -> load:0x40078000,len:10900
01:19:21.888 -> load:0x40080400,len:6360
01:19:21.888 -> entry 0x400806b4
01:19:23.393 -> Waiting for WiFi to connect.... connected
01:19:24.381 -> Waiting for NTP time sync: 
01:19:24.381 -> Current time: Fri Sep 24 08:19:33 2021
01:19:25.696 -> Found city, woied:
01:19:25.696 -> 2436704

brandonroots, 2022-10-04:
Found a solution [here](https://github.com/esp8266/Arduino/issues/2281#issuecomment-258706478) for the conflict between File class defined in FS.h and SD that does not require any changes to the Inkplate library. 

As shared by [luc-github](https://github.com/luc-github) there is a define for this to be used before header call in the Arduino sketch:
`#define FS_NO_GLOBALS
#include <FS.h>`

then you must add namespace fs for SPIFFS objects like
`fs::File currentfile = SPIFFS.open("/myfile.txt", "r");`
but for SD no need :
`File textfile = SD.open("/myfile.txt", FILE_READ);`

This resolves the conflict

TheWebMachine, 2023-03-08:
> Found a solution [here](https://github.com/esp8266/Arduino/issues/2281#issuecomment-258706478) for the conflict between File class defined in FS.h and SD that does not require any changes to the [...]

Thank you for this! It ended up being the first useful result in my search for an answer to this problem.

KarloLeksic, 2023-04-13:
The issue is closed because there is no more activity

closed 2023-04-13 by KarloLeksic