SolderedElectronics/Inkplate-Arduino-library #102: Inkplate-problem with partial update and display function

ifadiga, 2021-09-16:
There is the topic (https://forum.e-radionica.com/en/viewtopic.php?f=23&t=395) with issues where have a difference on the picture when using display.display and display.partialUpdate.
With partialUpdate I have a better picture, while with display.display picture is brighter.

michaelotto, 2021-09-16:
Here is some sample code that demonstrates the issue. The image looks much different (faint, low contrast) in the "After" part in comparison to the "Before" part.

`#include <WiFi.h>                   // Include ESP32 WiFi library
#include "Inkplate.h"               // Include Inkplate library to the sketch

#define WIFI_SSID                   "..."
#define WIFI_PSK                    "..."

Inkplate display(INKPLATE_1BIT);            // Create an object on Inkplate library and also set library into 1 Bit mode (Monochrome)

void setup() {
  Serial.begin(115200);
  
  if (WiFi.SSID() != WIFI_SSID) {
    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_SSID, WIFI_PSK);
  }

  while (WiFi.status() != WL_CONNECTED)
    delay(500);

  Serial.println("Wifi connected");
  
  display.begin();                          // Init Inkplate library
  display.clearDisplay();
  display.setRotation(1);
  display.display();

  // Before part
  display.drawImage("http://el-otto.de/display10.png", 0, 0);
  display.partialUpdate();

  delay(15000);

  // After
  display.display();  
}

void loop() {}`

michaelotto, 2021-10-03:
I have some more observation on this:

display.display() behaves the same way as if all the pixels were turned to black beforehand.

Look at the following example

```
#include <WiFi.h>                   // Include ESP32 WiFi library
#include "Inkplate.h"               // Include Inkplate library to the sketch

#define WIFI_SSID                   "..."
#define WIFI_PSK                    "..."

Inkplate display(INKPLATE_1BIT);            // Create an object on Inkplate library and also set library into 1 Bit mode (Monochrome)

void setup() {
  Serial.begin(115200);
  
  if (WiFi.SSID() != WIFI_SSID) {
    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_SSID, WIFI_PSK);
  }

  while (WiFi.status() != WL_CONNECTED)
    delay(500);

  Serial.println("Wifi connected");
  
  display.begin();                          // Init Inkplate library
  display.clearDisplay();
  display.setRotation(1);
  display.display();

  display.drawImage("http://el-otto.de/display10.png", 0, 0);
  display.partialUpdate();

  delay(15000);

  display.fillRect(0, 0, 825, 1200, BLACK);
  display.partialUpdate();  
  display.drawImage("http://el-otto.de/display10.png", 0, 0);
  display.partialUpdate();  
}

void loop() {
}
```

It comes to the same "washed out", "low contrast", "light" image by turning all the pixels to black and then displaying the image again with partialUpdate.

BornaBiro, 2021-11-16:
We are a little bit late on this one, we are really sorry about that, but we think we found a problem.
If you have problems with image fading over time, picture being too dark or with very high contrast, or with severe ghosting you could try light mode in the new version of library for Inkplate 10 boards.
You can find more details here: https://forum.e-radionica.com/en/viewtopic.php?f=23&t=407&p=810#p810

michaelotto, 2021-11-16:
Thanks. Does this only refer to grayscale mode or also 1-bit? (Because I'm using 1bit only). BTW: You forgot to bump up the version number in the 5.5.0 release ;-)

nitko12, 2022-02-10:
Yep, 

should work with 1 bit mode too, I'm closing stale and resolving issues now, so I'll close this, feel free to reopen if you need anything.

Thanks for the contribution :)

closed 2022-02-10 by nitko12