SolderedElectronics/Inkplate-Arduino-library #104: Feature suggestion: partialUpdate should return how many pixels were updated

michaelotto, 2021-09-26:
partialUpdate, which is void at the moment, should return how many pixels were updated/changed. This way, one could better decide how much need there is for a full refresh.

I looked at the code and figured this would be a not too difficult change. I could try myself and provide a pull request if you like.

michaelotto, 2021-09-26:
This would be the snippet where the counting takes place. It counts only pixel changes from white to black as these cause visible blur.

```
    uint32_t changeCount = 0;

    for (int i = 0; i < E_INK_HEIGHT; ++i)
    {
        for (int j = 0; j < E_INK_WIDTH / 8; ++j)
        {
            diffw = *(DMemoryNew + _pos) & ~*(_partial + _pos);
            diffb = ~*(DMemoryNew + _pos) & *(_partial + _pos);
            if (diffw)                                                      // count pixels turning from black to white as these are visible blur
            {
                for (int bv = 1; bv < 256; bv <<= 1)
                {
                    if (diffw & bv)
                        changeCount++;
                }
            }
            _pos--;
            *(_pBuffer + n) = LUTW[diffw >> 4] & (LUTB[diffb >> 4]);
            n--;
            *(_pBuffer + n) = LUTW[diffw & 0x0F] & (LUTB[diffb & 0x0F]);
            n--;
        }
    }

```

nitko12, 2021-09-27:
I've added your code to the dev branch for all displays here:

https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/8ba09f8ec78548e1842ea5c8bcb7ee257b00bb1f

It will be merged to master when some more features are added.

Thanks for the suggestion :)

closed 2021-09-27 by nitko12