SolderedElectronics/Inkplate-Arduino-library #111: Help needed to plug PS/2 keyboard

dedesite, 2021-11-11:
Hello,
I try to build a e-ink typewriter with my Inkplate 6 but I'm very new to micro controller development and electronics in general.
I already tried to use a BT Keyboard but for now there is a bug when board works as master (see #98 ).

Si, I try to use an old PS/2 with the [PS2KeyAdvance library](https://github.com/techpaul/PS2KeyAdvanced).
On the readme, it stated two things that I don't fully understand : 

> Make sure you have data and clock wires connected correctly. Clock wire MUST be connected to an interrupt pin.

What is an interrupt pin ? Is there one on the Inkplate 6 ?

> When using DUE or other boards with 3V3 I/O you MUST use a level translator FET or IC like Texas Instruments TXS0102 or similar as most keyboards not only operate at 5V but the two wire communications are pulled up by a resistor to 5V at the keyboard end.

I assume I need to use such a level translator since the board got 3.3V I/O, but I don't know how to plug it...

In the [library examples](https://github.com/techpaul/PS2KeyAdvanced/blob/82e13bbd79f045e1b7716a44b52a7f51fe1cc708/examples/SimpleTest/SimpleTest.ino#L134) they use numbers for the pins, but how to I know which number correspond to GPIO pins ? Should I use the MCP23017 pins or the ESP32 ones ?

Hope Github is the place to ask those questions :). Maybe forum is more suited ?

Thanks in advance for your help,
Andréas

BornaBiro, 2021-11-11:
Hello Andréas!

[e-radionica forum](https://www.forum.e-radionica.com/en/viewforum.php?f=20) is a much better choice for this kind of questions, but since you ask here, I will answer you.

Unfortunately, we could not help you with PS2KeyAdvance library. If you have any questions/problems about that library, the best option is to ask the author of that specific library.

If you don't know what interrupt pin is or interrupts in general, you can find some info [here](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/) and [here](https://www.allaboutcircuits.com/technical-articles/using-interrupts-on-arduino/) but these are general tutorials that are not strictly associated with Inkplate boards.

If you want to connect 5V device on Inkplate board (that runs on 3.3V) you must use some kind of voltage level converter/translator.

You can find all connections in our [schematic diagram](https://github.com/e-radionicacom/Inkplate-6-hardware/tree/master/Schematics%2C%20Gerber%2C%20BOM/v1.0)
Bear in mind that MCP23017 is I2C Input/Output expander, so it's not fast as ESP32 GPIOs. But also, Inkplate doesn't have many GPIO left on ESP32, because many of them are used by e-paper panel.

BornaBiro, 2021-11-16:
Because the are no further comments, I will close this issue.

dedesite, 2021-11-17:
Sorry for the late answer,  thank you for your reply @BornaBiro, I bought a voltage level converter, we will see how it goes with a PS2 keyboard :).

dedesite, 2021-12-12:
Hi @BornaBiro,
I continue working on PS/2 keyboard support since porting USB Host to Inkplate is not that easy... I think I found what to do, but got stuck in usb initialization and don't why...
Can you confirm me that the INT pin to use on Inkplate is the on labeled "INTA" on the MCP23017, or should I use another pin ?
BTW, I also had to buy a DC-DC step up converter since a level converter doesn't seem to boost the voltage, only reduce it to 3.3V.

Andréas