SolderedElectronics/Inkplate-Arduino-library #112: Inkplate 10, washed out colors

Stanko, 2021-11-15:
First - thank you a lot for the making Inkplate, it really helps starting with eink projects!

Although, I do have problem with mine, and I'm not sure if I'm doing something wrong. It seems like grayscale colors are not working, and lines are super washed up.  If you check the image below, you'll see that the parts of the line are differently colored.

What I noticed is that it draws the line in solid black and then it fades away. Black works well for filled shapes (circles for example), but shades of gray do not work for fill wither.

I'm using the latest arduino library from this repo (downloaded master branch as zip).

### Photo:

[image]

### Video of the initial refresh:

https://user-images.githubusercontent.com/776788/141816921-7934bea7-34a6-4da6-a6c2-2b5b09dc132d.MOV

### Code used:
```c
#include "Inkplate.h" //Include Inkplate library to the sketch
Inkplate display(INKPLATE_3BIT);

void setup()
{
    display.begin();        
    display.clearDisplay(); 
    display.display();      
}

void loop()
{
    display.drawCircle(100, 100, 50, 0); 
    display.drawCircle(200, 100, 50, 1); 
    display.drawCircle(300, 100, 50, 2); 
    display.drawCircle(400, 100, 50, 3); 
    display.drawCircle(500, 100, 50, 4); 
    display.drawCircle(600, 100, 50, 5); 
    display.drawCircle(700, 100, 50, 6); 
    display.drawCircle(800, 100, 50, 7); 
    
    display.fillCircle(100, 200, 50, 0); 
    display.fillCircle(200, 200, 50, 1); 
    display.fillCircle(300, 200, 50, 2); 
    display.fillCircle(400, 200, 50, 3); 
    display.fillCircle(500, 200, 50, 4); 
    display.fillCircle(600, 200, 50, 5); 
    display.fillCircle(700, 200, 50, 6); 
    display.fillCircle(800, 200, 50, 7); 
    
    display.display();

    delay(100000);
}
```

## Noise

Also sometimes there is some weird noise showing up:
[image]

Stanko, 2021-11-15:
I just tested this image:
https://i.imgur.com/3XWOATW.png

And this is the result (the bottom part is with dithering on). This is not the result I expected, does this mean my display is faulty?

[image]

```c
display.drawImage("https://i.imgur.com/3XWOATW.png", 0, 0, 0, 0);
display.drawImage("https://i.imgur.com/3XWOATW.png", 0, 400, 1, 0);
```

BornaBiro, 2021-11-16:
Hello Stanko.

You could try use Inkplate 10 Light mode by writing display.begin(1); instead of display.begin();
It will use a different waveform, so it should make the picture brighter.

Stanko, 2021-11-16:
It definitely helped with grays, but still, lines are still almost unusable. Also I got these uneven patches when I tried the gradient image again.

[image]

[image]

BornaBiro, 2021-11-16:
There can be a problem with VCOM voltage on your panel.
Try setting up a new VCOM voltage following [this procedure](https://forum.e-radionica.com/en/viewtopic.php?f=23&t=407&p=810#p810) while still using light mode.

I hope this will fix the problem.

Stanko, 2021-11-16:
Great! I will try and report back. Thanks again.

Stanko, 2021-11-16:
I think I managed it, and these are the results I'm getting:

Is there a way I can check if I was able to set the voltage to -1.1?

And sorry for being a nuisance 🙈 

[image]
[image]

BornaBiro, 2021-11-16:
Panel has refreshed and showed splashscreen, which means VCOM programming is successful.

As far as I can see in this photo, the results are now much better. You have to bear in mind that Inkplate library uses one (or in this case two) waveform LUTs that has to fit for every Inkplate 10 board, which means that a perfect grayscale gradient is not possible on every board. Also using dither can also create such weird effects on color change, so you can turn dither off, but then you can only use 3 bit grayscale.

Stanko, 2021-11-16:
Not sure if it can be seen on the picture, but the result is pretty much the same as before setting the voltage. 

Well, this is very unfortunate:
* I can't draw black lines, they are still washed out (photo below)
* Color of filled shapes is not uniform (you can still see uneven patches on the photo above)
* Scale of colors is also not uniform making the display pretty much only usable in black and white mode (any photo I tried looked pretty bad).

What do you think, is this  a hardware or a software issue? I would love to get better gray scale mode, because that was the main reason I got inkplate 10.

[image]

BornaBiro, 2021-11-17:
> I can't draw black lines, they are still washed out (photo below) 

You can try to draw thicker lines and see is there any difference.

> Color of filled shapes is not uniform (you can still see uneven patches on the photo above)

> Scale of colors is also not uniform making the display pretty much only usable in black and white mode (any photo I tried looked pretty bad).

I'm sorry to hear that, but this is a technical limitation of Inkplate. It can draw grayscale 3 bit, but because the whole Inkplate is based on reverse engineering of commercial epaper products like ebook readers and also it uses refurbished epaper panels, the results won't be perfect.
This is even stated on [Inkplate 10 Campaign page](https://www.crowdsupply.com/e-radionica/inkplate-10).

Stanko, 2021-11-17:
Well, it is what it is, I'll try to make the best of it. Thank you very much for the info and patience!

closed 2021-11-17 by Stanko

Stanko, 2021-11-17:
And I just want to mention, it is still brilliant that you guys created something like this, I was just hoping for more colors 🙂 

To be fair, black and white imagery looks amazing (and the most of my generative art is black and white anyway):

[image]

[image]

[image]

davidzovko, 2021-11-17:
This really looks stunning - thanks for sharing!