SolderedElectronics/Inkplate-Arduino-library #114: Bug with getTextBounds and GFX Fonts ?

dedesite, 2021-11-18:
Hi,
I've been trying to use `getTextBounds` on only one characters with a GFX fonts, and its seems to miscalculate the width of the letters (bug in `Adafruit_GFX::charBounds` ?)

Here is a code, that display all the alphabet and letter bounding box letter with default font and GFX font.
You will notice, that sometimes, the letter is not entirely in the bounding box, and that there is sometimes space between bounding boxes.

```c
#include "Inkplate.h"

#include "Fonts/Open_Sans_SemiBold_32.h"

Inkplate display(INKPLATE_1BIT);

const uint8_t color_black = 1;
const GFXfont *text_font = &Open_Sans_SemiBold_32;
const int ascii_upper_shift = 32;

void display_letter_bounding_box(String letter)
{
  // debug
  int16_t cursor_x = display.getCursorX();
  int16_t cursor_y = display.getCursorY();
  int16_t x, y;
  uint16_t w, h;
  display.getTextBounds(letter, cursor_x, cursor_y, &x, &y, &w, &h);
  display.drawRect(cursor_x - w, y, w, h, color_black);
}

void print_alphabet(bool upperCase)
{
  char alphabet[27] = "abcdefghijklmnopqrstuvwxyz";
  for (int i = 0; i < 26; i++)
  {
    String letter_str = String(alphabet[i]);
    if (upperCase)
    {
      letter_str.toUpperCase();
    }
    display.print(letter_str);
    display_letter_bounding_box(letter_str);
  }
}

void setup()
{
  display.begin();
  display.clearDisplay();
  display.setCursor(0, 50);

  display.setTextSize(3);
  print_alphabet(false);
  display.println("");
  print_alphabet(true);

  display.println("");
  display.println("");
  display.println("");
  display.println("");
  display.setFont(text_font);
  display.setTextSize(1);

  print_alphabet(false);
  display.println("");
  print_alphabet(true);

  display.display();
}

void loop()
{
  delay(200);
}
```

If I can help fixing this, I would be very glad.
Thanks,
Andréas

BornaBiro, 2021-11-22:
You have a few errors in the code. First one is

```cpp
display.getTextBounds(letter, cursor_x, cursor_y, &x, &y, &w, &h);
display.drawRect(cursor_x - w, y, w, h, color_black);
```

You need to pass x and y form getTextBounds() to drawRect(), so it actually should be

```cpp
display.getTextBounds(letter, cursor_x, cursor_y, &x, &y, &w, &h);
display.drawRect(x, y, w, h, color_black);
```

Second bug is this one

```cpp
display.print(letter_str);
display_letter_bounding_box(letter_str);
```

As soon as you print something, your cursor has moved. And because you are getting cursor position in display_letter_bounding_box() to write bounding box, you are writing it in wrong position.
What you should do is bounding box first and then print it.

```cpp
display_letter_bounding_box(letter_str);
display.print(letter_str);
```

dedesite, 2021-11-22:
Thank for your reply @BornaBiro, I can't test right now, but as I understand your remark and my code, I think it would behave the same, but I may be wrong.

The two bugs your are pointing are relate and it is because I call `display_letter_bounding_box` after the `print` statement that I used `cursor_x-w` instead of `x` in the `drawRect`.

So as I understand it, it's not a bug, just a different way to do the rendering.
Will try as I can though and see if it change something :).

dedesite, 2021-12-24:
@BornaBiro You were right, I just tried and it works, thanks ! ;)

BornaBiro, 2022-01-04:
No problem! :)
I will close this issue for now.

closed 2022-01-04 by BornaBiro