SolderedElectronics/Inkplate-Arduino-library #115: Platformio setup

makeworld-the-better-one, 2021-11-21:
In #16 @martinberlin shared a repo for using Inkplate with Platformio: https://github.com/martinberlin/inkplate-platformio

That repo is out of date though, and doesn't support devices like the Inkplate 10. Is there any chance we could get an official platformio config? Thanks!

jakobwesthoff, 2021-11-21:
I don’t know if anything official, but you can use the code from my first tinkering with Inkplate 10 some time ago as a starting point. I used Platform.it

https://github.com/jakobwesthoff/inkplate10-first-tinkering

makeworld-the-better-one, 2021-11-21:
Ooh looks good thanks! 

https://github.com/jakobwesthoff/inkplate10-first-tinkering/blob/79c3d8735e63de2aabe3cb6af0f6e06535bb7f51/platformio.ini#L29

Pretty sure that line is no longer needed, if you click the lib you'll see it's deprecated.

I'll try this out.

jakobwesthoff, 2021-11-21:
Yeah you are right, there might be quite some stuff, which is not needed any more or can be done better. As I said I did this quite some time ago ;). However as a starting point it should be a nice thing. Maybe I will create a real inkplate10 platform.io starter repo in the future.

nitko12, 2022-02-07:
I've created a port of the library to PlatformIO, it can be found in registry or here https://github.com/e-radionicacom/Inkplate-Platformio-Library ,

but as for the board definition, you can find JSONs here, as they haven't yet added them as boards.

https://github.com/e-radionicacom/platform-espressif32
(here's a PR to espressif platformio https://github.com/platformio/platform-espressif32/pull/698 too)

closed 2022-02-07 by nitko12