SolderedElectronics/Inkplate-Arduino-library #116: Unnecessary RTC calls

makeworld-the-better-one, 2021-11-23:
The RTC example has code like this:

```cpp
    display.rtcReadTime();             // Get the time and date from RTC
    seconds = display.rtcGetSecond();  // Store senconds in a variable
    minutes = display.rtcGetMinute();  // Store minutes in a variable
    hour = display.rtcGetHour();       // Store hours in a variable
    day = display.rtcGetDay();         // Store day of month in a variable
    weekday = display.rtcGetWeekday(); // Store day of week in a variable
    month = display.rtcGetMonth();     // Store month in a variable
    year = display.rtcGetYear();       // Store year in a variable
```

But `System.cpp` has code like this for all the `rtcGet` methods:

```cpp
uint8_t System::rtcGetSecond()
{
    rtcReadTime();
    return rtcSecond;
}
```

It seems it's calling `rtcReadTime` every time, which is inefficient, and unnecessary with the example code anyway. Could it be removed? This is also good to get one consistent date time. Consider the case where you call `rtcGetDay`, but by the time you get around to calling `rtcGetMonth` it's past midnight and the month has changed. This would result it an inconsistent date time.

makeworld-the-better-one, 2021-11-23:
It would also be nice if there was an `rtcGetTm` function or something that returns a `time.h` `tm` type.

BornaBiro, 2021-11-29:
Yes, you are right. This is absolutely bad from a performance standpoint. We missed that.
There will be a function that will get the RTC data, (something like rtcReadTime()) but it will store that data internally in ESP32. Functions like rtcGetSecond(), rtcGetMinute() etc will only return that read data.

It returns epoch with [rtcGetEpoch()](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/include/System.h#L132) so you can easily convert it into struct tm with localtime() or gmtime().

makeworld-the-better-one, 2021-11-29:
Good to hear, thanks!

> (something like rtcReadTime()) but it will store that data internally in ESP32

Isn't this what `rtcReadTime` does already? Or do you mean something different.

> It returns epoch with rtcGetEpoch() so you can easily convert it into struct tm with localtime() or gmtime().

Yes, but `rtcGetEpoch` already constructs a `struct tm`, then converts to epoch. To convert back to a struct is possible of course, but inefficient. It'd be nice if there was a function that just returned the struct.

makeworld-the-better-one, 2022-02-09:
Looks like this was fixed by https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/483bd39d2c709d69a98cbf4a52310028e2f5cd5e, which is in the recent 5.6.0 release. Thanks @gogi640!

closed 2022-02-09 by makeworld-the-better-one