SolderedElectronics/Inkplate-Arduino-library #118: Is this correct?

geiseri, 2021-11-28:
Hi, I am working on updating the ESPHome code for the inkplate10, and I noticed this line:
https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/1f8bf43bfc659fe7e11812a363027c1e7da48645/src/boards/Inkplate10.cpp#L38

Is this intentional?

BornaBiro, 2021-11-29:
Yes. But can you be more specific about the problem?
This is the second waveform for Inkplate 10 if you having issues with a too dark refresh, burn-in, or fading/darkening like [this](https://forum.e-radionica.com/en/viewtopic.php?f=23&t=407).

geiseri, 2021-11-29:
I was confused since the memory it was being copied into was defined as [WAVEFORM3BIT_LIGHT ]( https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/1f8bf43bfc659fe7e11812a363027c1e7da48645/src/boards/Inkplate10.h#L89) and that is a `uint8_t[8][8]`. I don't think it's fatal, but I think it might have garbage data in there.

nitko12, 2022-02-10:
https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/01472690641ff0c53c06336dcc156b5ac4abd1ba

Here's the fix, thanks :)