SolderedElectronics/Inkplate-Arduino-library #122: mcp interrupt registers are cleared before they can be read.

lanrat, 2021-12-14:
The `display.begin()` method initializes the i2c bus for talking to the MCP23017 and then [immediately clears the MCP's registers](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/src/boards/Inkplate10.cpp#L47) before allowing any other method to read them out.

This is a problem if the MCP's registers contain information needed before being() is called.
For example, if setting the MCP's interrupts to wake the ESP32 on a touchpad event, the MCP can wake the ESP32 but once woken, the ESP has no way of knowing which touchpad was responsible for the wake event as that register is cleared in `begin()`.

I have been able to hack around this by manually initializing the i2c bus, reading the MCP registers, and initializing the i2c bus, and finally calling `display.begin();`

However, this solution is hacky and non-idea. I propose saving the MCP's registers in `begin()` before they are cleared, and then allow them to be read out with a getter function.

This would allow developers to ask the display library about the registers at wake time and know which touchpad events (or anything really) caused the wake event.

If this sounds like a good idea I'm happy to provide a PR.

urb4nc4rl, 2022-01-07:
I personally think that this is a good idea and would have use of this functionality.  (to know which touchpad woke the device from sleep) -- I would be very grateful for the PR and happy to test it.

nitko12, 2022-02-10:
We're sorry, 

but removing those would result in a terrible low power mode, which is a more common use case.

A workaround like yours is a plausible option for advanced users tho, great idea :)

closed 2022-02-10 by nitko12

nitko12, 2022-02-14:
Hi,

this sorry if I understood you wrong,
we'll add this in the future probably, if you find time and will tho you could do a PR with proposed addition to dev branch :)