SolderedElectronics/Inkplate-Arduino-library #123: How to use interrupt pins ?

dedesite, 2021-12-16:
Hi,
I've been trying to build a electronic typewriter with an Inkplate 6 for the last 2 mounths.
It's been a pleasant experience, but I'm a bit stuck on getting keyboard to work.

I now try to get PS2 keyboard working with [PS2KeyAdvanced library](https://github.com/techpaul/PS2KeyAdvanced) that I need to modify to use the MCP extender.

I have to plug the data wire on a GPIO (I'll certainly use GPB5) and the clock wire to an interrupt pin. My understanding is that "INTA" is the right one.

Then, I just need to change the call to `pinMode` and `digitalWrite` to the MCP equivalent `pinModeMCP` and `digitalWriteMCP`.

But is there a equivalent for `attachInterrupt` and `digitalPinToInterrupt` on the MCP ?
Is there other interrupt pin available on the ESP32 ?

Have you other info, doc on how to use interrupt pins with Inkplate ?

Thanks,
Andréas

P.S : I know the forum should be a bette place to post this kind of question, but can't register :(

lanrat, 2022-01-05:
There is a way to do this, but its not exposed very well in the existing Inkplate API.

The MCP has two interrupt lines that you can have the ESP attach to, and you can tell the MCP which pins will trigger the interrupt to alert the ESP. Then in the ESP interrupt you can read the interrupt registers from the MCP to see which of its pins triggered the interrupt sent to the ESP. I recommend reading the MCP datasheet to fully understand how this works.

You can find a similar example in my PR https://github.com/e-radionicacom/Inkplate-Arduino-library/pull/121/files which does exactly this to determine which touchpad on the MCP triggered the interrupt.

nitko12, 2022-02-07:
Here's an example which sets interrupts for waking up from deep sleep on the MCP.

https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/master/examples/Inkplate6/Advanced_Inkplate_Features/Inkplate_Wake_Up_On_Touchpads/Inkplate_Wake_Up_On_Touchpads.ino

You could make it run a callback instead of waking up if that's what you need.

If you need anything else feel free to reopen the issue :)

closed 2022-02-07 by nitko12