SolderedElectronics/Inkplate-Arduino-library #124: Disable Front on Boot Inkplate 6plus

degan6, 2021-12-18:
Hello, Thank you for making these awesome products and software to go with it.

I am wondering if it would be possible to update the boot sequence to default the backlight to off. Right now it comes on at full brightness after calling ` display.begin()`. I can quickly set it off but this results in the backlight flashing on then quickly off.

So the end result would be that the backlight never comes on from boot unless `display.frontlight(true);` is called.

Thank you!

If this is already possible let me know

Critical-Impact, 2022-01-12:
Would love to see this functionality added as well, using my inkplate as a home display and I have it in a cycle of sleeping/waking up and as such I would like the backlight to stay off.

itobey, 2022-01-19:
Linking the thread of the eradionica forum: https://forum.e-radionica.com/en/viewtopic.php?f=22&t=417
Looking forward to the merge!

itobey, 2022-01-22:
Bumping this again since they have fixed the issue but not published a new version of the lib yet. If you want to play around with it, just do this change on your local file of the lib: https://github.com/e-radionicacom/Inkplate-Arduino-library/commit/b59fb4b564294afd2c57edda7097724f1254ba55
I can confirm that after setting this to `LOW` my Inkplate 6 plus keeps the frontlight off.

freman, 2022-01-24:
Perhaps this could be a conditional define that can be overwritten at build time - sort of a per project thing? I've definitely got some projects where I'd rather not have the screen flash on while rebooting (or waking from sleep) and some where I'd rather not have it flash off.

nitko12, 2022-02-10:
Hi, 

adding a flag would probably be something not many beginners would use, I'd say that having it off by default is the way to go, I'ts easy to turn it on manually after boot :)

I'll close this now, feel free to reopen if you need anything else.

closed 2022-02-10 by nitko12