SolderedElectronics/Inkplate-Arduino-library #127: Comment incorrectly states that enclosure doesn't support touchpad

urb4nc4rl, 2022-01-04:
https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/1f8bf43bfc659fe7e11812a363027c1e7da48645/examples/Inkplate10/Advanced_Inkplate_Features/Inkplate_Touchpads/Inkplate_Touchpads.ino#L18

There is a note in the touchpad example files for Inkplate 10 that says:

` NOTE: You can not use touch pads when enclosure is fitted on the Inkplate - they are not that sensitive!`

However, the [official Inkplate news and updates](https://www.crowdsupply.com/e-radionica/inkplate-6/updates/too-many-new-things-for-one-headline) states:

`Touch pads working through the 3D printed enclosure`

...I haven't tested yet, but I presume the code file is out-of-date and the comment in code needs to be removed?

lanrat, 2022-01-05:
The 3D print includes some [small blocks](https://github.com/e-radionicacom/Inkplate-10-hardware/blob/main/3D%20printable%20case/Original%20case/Inkplate%2010%20touch%20pads%20square.STL) that if you cover in foil and put in the touchpad's cavity in the case will allow the touchpads to work through the case.

I agree this should be made more clear.

nitko12, 2022-02-10:
Hi,

I've removed the note from the examples, it works in the case true, but not as reliable as without it, which probably can and should be improved.

Your workaround with the foil is certainly an option :)

closed 2022-02-10 by nitko12