SolderedElectronics/Inkplate-Arduino-library #129: Inkplate 6Plus Wake from Touchscreen?

freman, 2022-01-20:
Hi,

I've been poking around and even tried getting on the forum but haven't been able to get onto the forum, so I'll ask my question here.
Is it possible to wake from deep sleep with a touchscreen event?

mkocus, 2022-01-20:
Also would like to know that.

itobey, 2022-01-22:
They have provided an example for this, but not merged it to master yet. See this on the dev branch:
https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/dev/examples/Inkplate6PLUS/Advanced_Inkplate_Features/Inkplate_Wake_up_on_touchscreen/Inkplate_Wake_up_on_touchscreen.ino

nitko12, 2022-02-09:
Hi,

indeed there is an example, but it defeats the purpose of low power, as the touch controller doesn't have low power mode and will drain the battery pretty quickly, so watch out.

Also if I recall correctly, I couldn't get the coordinates after waking up, @freman have you managed to do that, if that's what you're trying to acheive?

closed 2022-05-31 by BornaBiro